pftf/RPi4 #2: Could I report issues on this project?  INTERRUPT EXCEPTION NOT HANDLED

ShireCiel, 2019-12-06:
Thanks for your works. It's amazing USB port works fine.
I don't know could i report problem on this "doesn't have main code" project.
If cann't, please forgive.
I try boot windows 10 and windows 10 PE on latest release efi fv.
Got stop code: INTERRUPT EXCEPTION NOT HANDLED.
I enable debug and bootlog in boot options. It not write any debug file to boot storage.
Looks it normally started load drivers. (I patched some unsigned drivers, it can't boot  and show crash on load xxx.sys before i am not yet add  "testsigning on" boot option).
In previous version of code. Thought cannot use USB on pi4, but It could boot to WIndows 10 PE normally(after boot up, scripts works fun, just cannot control on otg or usb).
Maybe PCI-E support cause some problem?
Can u fix it?

pbatard, 2019-12-06:
Again, this project is EXPERIMENTAL, which means it is contains boths unstable and unfinished code.

If you expect to be able to boot Windows with the current status of the project, you will only be disappointed.

Please do not ask for support about running Windows at this stage (or even vanilla Linux).

closed 2019-12-06 by pbatard

andreiw, 2019-12-06:
This is a work in progress. Windows is not supported right now so I’d rather we not start hyping up drama. When we claim Windows works you can file away actual issues.

A

Biswa96, 2020-01-30:
Is there a guide that I can follow to try this project to boot a arm Linux distribution or Windows?

andreiw, 2020-01-30:
No guide yet as it’s not yet ready for general consumption and many things don’t work.

A

Biswa96, 2020-02-09:
Is it possible to add a guide about build instructions? There are 5 build commands in build_firmware.sh file, do we need all of them to compile?

pbatard, 2020-02-09:
Sorry but I'm afraid this is just not the type of project where we are going to hold the hand of people who are unfamiliar with the edk2-platftorms build process.

Please bear in mind that this project is just an addon on top of [edk2-platforms](https://github.com/tianocore/edk2-platforms/) to produce binaries for people who aren't able to do that themselves. This means we either expect you to have enough know-how to make enough sense of the default documentation that comes from [edk2-platforms](https://github.com/tianocore/edk2-platforms/), or, if you can't, just use the binaries we publish precisely for the people who would like to build the firmware, but don't have enough expertise to know how to do so.

Adding extra documentation, that would mostly be a re-hash of what can already be found in edk2-platforms would be both counter-productive and go against the goal of this project, which is only to provide read-made binaries for end-user consumption.