pftf/RPi4 #5: arm64 debian installator doesn't see sd card / network devices

michalmilkowski97, 2020-02-23:
Hi,

first of all thank you for your work and support. The UEFI on rpi4 works well, however, I encountered several problems trying to install official debian arm64 os (using https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html as template for version for pi4)

The installator boots up normally but doesn't see sd card (no mmcblk entries in /dev), or ethernet / WiFi devices, so installation of system can only be done on pendrive using image from pendrive (because is'nt possible to mount sd card).

Do you have any idea what could cause that behaviour?

Biswa96, 2020-02-23:
The readme file already touches the issue, as below:

> with the caveat that you NIC and SD card will be unavailable due to the current Debian kernel missing up to date drivers

I have installed Debian in a USB drive, see my steps [here](https://github.com/Biswa96/rpi-backup). After all USB 3.0 interface will be faster than a normal mSD card.

michalmilkowski97, 2020-02-23:
Actually it is in readme, sorry my misread. I'll try your method. Thanks for reply.

michalmilkowski97, 2020-02-23:
So there is no way to have working NIC on rpi4 with debian arm64 now?

Biswa96, 2020-02-23:
> So there is no way to have working NIC on rpi4 with debian arm64 now?

See the last comment of this thread https://github.com/pftf/RPi4/pull/4, pbatard mentioned it.

closed 2020-02-23 by pbatard

andreiw, 2020-02-24:
NIC should be okay now as of 1.1. Everything else is still under very active development. Feel free to follow https://rpi4-uefi.dev for updates.