pftf/RPi4 #11: Validate that JTAG routing setting actually does something meaningful on Pi 4

andreiw, 2020-03-01:
The dirty secret was that it wasn't really tested on the Pi 3 either.

samerhaj, 2020-05-20:
Someone on my team is working on this. Will update with results soon.

PhatFree, 2020-06-04:
So as it sits it doesn't work, I have a commit that fixes.  Although from my investigations I'm not sure this does anything meaningful. as far as I can tell right now, this only works when you have `enable_jtag_gpio=1` in `config.txt` effectivelyeffectively making this a redundant option, as you have to have both set.

 Interestingly enough it will close down the JTAG access once UEFI is up and running, but its open until UEFI is up.

PhatFree, 2020-06-04:
I have read only access at the moment, so here is the patch... 
``` From 482b226e5b40785695f9c00a243e868d2d976a0b Mon Sep 17 00:00:00 2001
From: Jeff Booher-Kaeding <jeff.booher-kaeding@arm.com>
Date: Thu, 4 Jun 2020 16:24:52 -0500
Subject: [PATCH] Fixed JTAG pinout for pi3/4

Updated the pinout to match the one on the Pi4, tested with the pi4, not yet with p3 but data sheet looks the same.
---
 Platform/RaspberryPi/Drivers/ConfigDxe/ConfigDxe.c | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/Platform/RaspberryPi/Drivers/ConfigDxe/ConfigDxe.c b/Platform/RaspberryPi/Drivers/ConfigDxe/ConfigDxe.c
index ac1004fe18..6e793efb82 100644
--- a/Platform/RaspberryPi/Drivers/ConfigDxe/ConfigDxe.c
+++ b/Platform/RaspberryPi/Drivers/ConfigDxe/ConfigDxe.c
@@ -502,7 +502,7 @@ ApplyVariables (
    * 1           VREF        N/A               1
    * 3           nTRST       GPIO22    ALT4    15
    * 4           GND         N/A               9
-   * 5           TDI         GPIO4     ALT5    7
+   * 5           TDI         GPIO26    ALT4    37
    * 7           TMS         GPIO27    ALT4    13
    * 9           TCK         GPIO25    ALT4    22
    * 11          RTCK        GPIO23    ALT4    16
@@ -510,14 +510,14 @@ ApplyVariables (
    */
   if (PcdGet32 (PcdDebugEnableJTAG)) {
     GpioPinFuncSet (22, GPIO_FSEL_ALT4);
-    GpioPinFuncSet (4, GPIO_FSEL_ALT5);
+    GpioPinFuncSet (26, GPIO_FSEL_ALT4);
     GpioPinFuncSet (27, GPIO_FSEL_ALT4);
     GpioPinFuncSet (25, GPIO_FSEL_ALT4);
     GpioPinFuncSet (23, GPIO_FSEL_ALT4);
     GpioPinFuncSet (24, GPIO_FSEL_ALT4);
   } else {
     GpioPinFuncSet (22, GPIO_FSEL_INPUT);
-    GpioPinFuncSet (4, GPIO_FSEL_INPUT);
+    GpioPinFuncSet (26, GPIO_FSEL_INPUT);
     GpioPinFuncSet (27, GPIO_FSEL_INPUT);
     GpioPinFuncSet (25, GPIO_FSEL_INPUT);
     GpioPinFuncSet (23, GPIO_FSEL_INPUT);
-- 
2.17.1
```

cleverca22, 2020-06-04:
at least on the rpi1-3 line (not confirmed on a 4) a special register has to be written to before the arm is enabled:
https://github.com/librerpi/rpi-open-firmware/blob/022380bd291ad6588aab11a9aa822c9eb81d30b8/firmware/drivers/BCM2708ArmControl.cc#L223-L224

so its not really feasible to enable jtag after booting, only to modify the config and reset

PhatFree, 2020-06-04:
Thanks for the info, that makes sense. So until UEFI can't reasonably set this, this function really can just act as something to switch off the JTAG GPIO. 

I tested this with OpenOCD and a DP Bus-Blaster.

The pinout: 
[image]

andreiw, 2020-06-05:
Hmmm it may be possible via mbox (there is a read and write reg mechanism there now). Would need to check...

A

PhatFree, 2020-06-11:
I don't know much about setting the register via mailbox. but I'd be happy to work with someone to try and get things working in a more user friendly way.

In the mean time I have a branch with the change to make JTAG "workable": https://github.com/pftf/edk2-platforms/tree/pi4_dev_JTAG
It will still require making sure `enable_jtag_gpio=1` is in the `config.txt` 

Also below is a OpenOCD config that should work with a pi4, its set up for SMP, so you should be able to use it to debug all the way to the kernel.

```
transport select jtag

reset_config trst_and_srst

adapter_khz 1000
jtag_ntrst_delay 500

if { [info exists CHIPNAME] } {
  set _CHIPNAME $CHIPNAME
} else {
  set _CHIPNAME rpi4
}

if { [info exists DAP_TAPID] } {
   set _DAP_TAPID $DAP_TAPID
} else {
   set _DAP_TAPID 0x4ba00477
}

if { [info exists BOOTCORE] } {
   set _bootcore $BOOTCORE
} else {
   set _bootcore 0
}

if { [info exists CORES] } {
   set _cores $CORES
} else {
   set _cores 4
}

jtag newtap $_CHIPNAME tap -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_DAP_TAPID -enable
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.tap

set _TARGETNAME $_CHIPNAME.a72
set _CTINAME $_CHIPNAME.cti
set _smp_command "target smp "

set DBGBASE {0x80410000 0x80510000 0x80610000 0x80710000}
set CTIBASE {0x80420000 0x80520000 0x80620000 0x80720000}

for { set _core 0 } { $_core < $_cores } { incr _core } {

    cti create $_CTINAME.$_core -dap $_CHIPNAME.dap -ap-num 0 \
        -ctibase [lindex $CTIBASE $_core]

    set _command "target create $_TARGETNAME.$_core aarch64 \
          -dap $_CHIPNAME.dap -coreid $_core \
          -dbgbase [lindex $DBGBASE $_core] -cti $_CTINAME.$_core "
    set _command "$_command -rtos hwthread"      
    set _smp_command "$_smp_command ${_TARGETNAME}.$_core"

    eval $_command

    $_TARGETNAME.$_core configure -event reset-assert-post "aarch64 dbginit"
}

eval $_smp_command
```

samerhaj, 2020-06-19:
Any objection to submitting this (https://github.com/pftf/edk2-platforms/commit/482b226e5b40785695f9c00a243e868d2d976a0b) fix for now... while @andreiw is still looking at https://github.com/pftf/RPi4/issues/11#issuecomment-639692487 ?

samerhaj, 2020-09-23:
Patch submitted by @PhatFree https://edk2.groups.io/g/devel/message/65232

jlinton, 2020-09-29:
As a TODO, we need to assure that the JTAG pins are "disabled" if secure boot is enabled. So the pin routing "bug" can be redirected there.

cleverca22, 2020-09-29:
@andreiw not really, i believe i checked before, and attempting to enable jtag after the arm starts has no efect
it must be done before the arm has run a single opcode, which puts it purely in the domain of the VPU firmware

samerhaj, 2020-11-16:
the patch from https://edk2.groups.io/g/devel/message/65232 has been included in v1.21 FW Release. 

For the SecureBoot comment, I added a comment in #55 to track SecureBoot items

Closing this issue

closed 2020-11-16 by samerhaj