pftf/RPi4 #12: Clean-up BGRT handling in PlatformBootManagerLib

andreiw, 2020-03-01:
This bit of code existed in RaspberryPiPkg as a cute thing to do, to see the Windows logo on explicit launches of the Windows bootloader:

https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Library/PlatformBootManagerLib/PlatformBm.c#L719

This code doesn't work anyway because of missing upstream BootGraphicsResourceTableDxe patch (https://github.com/andreiw/RaspberryPiPkg/blob/3f5c70c9eb73e3f92ff42b05466f241561251b24/edk2Patches/0003-BootGraphicsResourceTableDxe-properly-handle-SetBoot.patch), so it could just be deleted (or we could spend time on edk2-devel arguing to get the DXE driver patch in). Anyway, for any kind of logo cert the OEM logo must be always shown...

There's no errant behavior seen...but the code is confusing, so I'll mark it as a v1.2 issue but not a bug.

andreiw, 2020-03-02:
Fix pushed by @ardbiesheuvel as f57b439e5610..30324eefb3a4

closed 2020-03-02 by andreiw

andreiw, 2020-03-02:
https://github.com/tianocore/edk2-platforms/commit/30324eefb3a4cadceb5afc022ca8a587eaed3453

pbatard, 2020-03-02:
Actually it's not in v1.3, coz that was built before Ard applied this commit.

pbatard, 2020-03-02:
Oh you mean, it's an issue that applies to 1.3  - my bad.