pftf/RPi4 #16: PlatformSmbiosDxe cleanup

andreiw, 2020-03-01:
Use PCDs and proper enums and defined structs (see Socionext\DeveloperBox\SmbiosPlatformDxe for an example)

Type 0: SystemBiosMajorRelease, SystemBiosMinorRelease, EmbeddedControllerFirmwareMajorRelease, EmbeddedControllerFirmwareMinorRelease are all required in SBBR 1.2, and Windows depend on them. We need reasonable values 

Type 1: UUID needs to be dynamic (per board, derived form SN and/or other board unique data). Added as an SBBR 1.2 requirement. Today this is derived from SN, but the SN is not unique enough. Should use MAC.

Type 2: Per SBBR, need pointer to Chassis (Type 3) and location in chassis.

Type 4: Missing SMCCC_ARCH_SOC_ID for CPU identification

Type 7: Invalid for Pi 4 (Pi 3 values)

Type 8: Per SBBR recommendation

Type 16: Wrong values for 2G and 4G Pi 4 SKUs (MaximumCapacity and ExtendedMaximumCapacity3)

Type 17: Wrong values (Size) for 2G and 4G Pi 4 SKUs and strings.

Type 41: Per SBBR recommendation

pbatard, 2020-03-10:
> Type 0: SystemBiosMajorRelease, SystemBiosMinorRelease, EmbeddedControllerFirmwareMajorRelease, EmbeddedControllerFirmwareMinorRelease are all required in SBBR 1.2, and Windows depend on them. We need reasonable values

We already have that. See `dmidecode` from the latest release (v1.5):
```
BIOS Information
        Vendor: https://github.com/pftf/RPi4
        Version: UEFI Firmware v1.5
        BIOS Revision: 1.5
        Firmware Revision: 20.2
```

The Firmware version is set to YY.M of `start4.elf` (here February 2020, which is about the best we can do when the Pi Foundation use an epoch timestamp). I upstreamed these changes in [tianocore/edk2-platforms@9369b01e86ad241f8807856bfc2ff64da7939876].

> Type 1: UUID needs to be dynamic (per board, derived form SN and/or other board unique data). Added as an SBBR 1.2 requirement. Today this is derived from SN, but the SN is not unique enough. Should use MAC.

How's the SN not unique enough __when one has been set__? It's a Serial Number.

Now, we do have RPi4 boards that effective report `0x0000000100000000` for the serial on the mbox query, but when that happens, we do use the MAC. So, once again, this part of the issue has been fixed some time ago. See [tianocore/edk2-platforms@180c613ee8af2824b1a8abc44652531816ca87bc].

I've added checkboxes to the list above so that we can remove the items that have already been addressed.

samerhaj, 2020-06-17:
There are multiple SMBIOS FWTS/ACS failures in Rel1.9:

https://github.com/pftf/acs-reports/blob/2ec70f92201f5f4ab36006ba36c500f3206b1958/v1.9/acs-enterprise-v1.6/luv-results-2020-04-21--14-08-49/parsed/fwts#L11

```
1.1  fwts FAILED INFO [HIGH] DMIInvalidHardwareEntry: Test 3, Incorrect Chassis Type SMBIOS
1.1  fwts Type INFO 3 reports 0x22
1.1  fwts ADVICE: INFO It may be worth checking against section 7.4 of the System Management
1.1  fwts BIOS INFO (SMBIOS) Reference Specification (see http://www.dmtf.org/standards
1.1  fwts /smbios). INFO /smbios).

1.1  fwts FAILED INFO [MEDIUM] DMIValueOutOfRange: Test 3, Invalid value 0x0000 was used and
1.1  fwts 0xffffshould INFO be used for non-PCI(e) while accessing entry 'System Slot
1.1  fwts Information INFO (Type 9)' @ 0x371d01e4, field 'Segment Group Number', offset 0x0d

1.1  fwts FAILED INFO [MEDIUM] DMIValueOutOfRange: Test 3, Invalid value 0x00 was used and
1.1  fwts 0xffshould INFO be used for non-PCI(e) while accessing entry 'System Slot Information
1.1  fwts (Type INFO 9)' @ 0x371d01e4, field 'Bus Number', offset 0x0f

1.1  fwts FAILED INFO [MEDIUM] DMIValueOutOfRange: Test 3, Invalid value 0x00 was used and
1.1  fwts 0xffshould INFO be used for non-PCI(e) while accessing entry 'System Slot Information
1.1  fwts (Type INFO 9)' @ 0x371d01e4, field 'Device/Function Number', offset 0x10
1.1  fwts ADVICE: INFO It may be worth checking against section 7.10 of the System Management
1.1  fwts BIOS INFO (SMBIOS) Reference Specification (see http://www.dmtf.org/standards
1.1  fwts /smbios). INFO /smbios).

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure Port Connector Information (Type 8) not found.
1.1  fwts Recommended INFO for platforms with physical ports.

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure OEM Strings (Type 11) not found. This
1.1  fwts structure INFO is recommended.

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure BIOS Language Information (Type 13) not found.
1.1  fwts This INFO structure is recommended.

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure System Event Log (Type 15) not found. This
1.1  fwts structure INFO is recommended.

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure IPMI Device Information (Type 38) not found.
1.1  fwts Required INFO for platforms with IPMI BMC Interface.

1.1  fwts SKIPPED: INFO Test 4, SMBIOS structure Onboard Devices Extended Information (Type 41)
1.1  fwts not INFO found. This structure is recommended.

1.1  fwts 24 INFO passed, 4 failed, 0 warning, 0 aborted, 6 skipped, 0 info only.
```

samerhaj, 2020-11-16:
There has been recent patches to EDK2 to create a re-usable generic SMBIOS library  code (for instance: https://edk2.groups.io/g/devel/message/67608). It would be much cleaner to use that new code . Closing this issue, and will open a seperate uefi-future issue to track future code change

closed 2020-11-16 by samerhaj