pftf/RPi4 #18: Document tools to burn SD card

vielmetti, 2020-03-01:
The README currently reads:

> Create an SD card in MBR mode with a single partition of type 0x0c (FAT32 LBA) or 0x0e (FAT16 LBA). Then format this partition to FAT.

it would be extraordinary helpful to document how this might be done with Linux, Mac, or Windows tools, perhaps in a separate section. E.g. "can I use `etcher` or `dd` to do this" is a reasonable request.

andreiw, 2020-03-02:
I believe this would be just formatting the card with legacy/MBR partition and a FAT32 partition.  In macOS:

To get here, click on the drive with SD card (not volume under drive), and select erase:

[image]

Note that if you don't see devices, just volumes, don't forget to:

[image]

Biswa96, 2020-03-02:
In Linux distributions, if you're using fdisk press `t` to choose type, press `c` for fat32. In Windows, right click on the drive, choose format, choose fat32 (default) option in the dialog box.

pbatard, 2020-03-02:
I believe this information is readily available for anyone willing to spend a couple minutes and therefore does not need to be added to the Readme.

The point of a Readme is to provide information that is pertinent to the specifics of the software being provided, with a specific demographics in mind. And, on a project such as this one, that demographics may not always be the one that caters for less tech-savvy users. 

This mean that providing handholding with regards to SD card formatting doesn't exactly qualify as the kind of elements that looks suitable for this project's Readme when it can easily be looked up externally, and we very much expect users of this project to be able to sort these kind of issues on their own.

I will therefore close this issue.

closed 2020-03-02 by pbatard

samerhaj, 2020-03-07:
I know this is closed and we did not feel including this info in Readme is useful. But I think at least some of the info mentioned here (https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md) is useful, especially for >32GB uSD cards