pftf/RPi4 #7: Support for external RTC.

andreiw, 2020-03-01:
Today the time/date is not kept across power state transitions. This is not a bug, because the Pi hardware has no RTC.

There are however Pi Hats with RTCs. It would be interesting to support such a hat (or hats). Of course, it needs to be done in way that doesn't break base Pi support.

TurnipG, 2020-05-12:
I have an i2c RTC module. Before using UEFI Firmware, I can add a udev rule that lets OS see the rtc device
After using UEFI Firmware, I found that UEFI Firmware will virtualize an rtc device (but the time is the same every time I boot ), and the /sys/class/i2c-adapter is empty, Raspberry Pi 4 UEFI Firmware Images does not support i2c yet?
If so, what do I need to do to get the OS to see i2c devices, such as config.txt?

andysan, 2021-02-16:
> I have an i2c RTC module. Before using UEFI Firmware, I can add a udev rule that lets OS see the rtc device
> After using UEFI Firmware, I found that UEFI Firmware will virtualize an rtc device (but [...]

You can't currently use I2C in Linux if you boot using UEFI. There are two reasons for this: 1) The kernel doesn't know how to attach the driver based on the information in the ACPI tables, 2) pin muxing isn't setup correctly. 

I have been able to use I2C from UEFI on a modified kernel with a modified firmware (I needed to initialise the pinmux from the firmware). Those patches aren't really in a state where I can share them since I had to hard-code the frequency of the clock driving the I2C block(s). I'm happy to clean them up and post them if I'm able to solve that issue though.

CoiaPrant, 2021-09-21:
My suggestion is that devices on I2C can be seen directly in Windows Device Manager, Linus is very simple to set up RTC, but Windows will be very difficult.

jlinton, 2022-06-02:
I hooked up the existing ds1307 driver, to the rpi's I2C1, and spent a whole bunch of time messing with depex/etc to get it to load automatically when enabled. The changes can be found here: https://github.com/jlinton/edk2-platforms/tree/cppc6-various-fixes-vc4aml-mailboxspinlock-xhcipci-clocktweaks-spiv5

The device I've been testing it with:
 https://www.amazon.com/Makerfire%C2%AE-Raspberry-Module-DS1307-Battery/dp/B00ZOXWHK4