pftf/RPi4 #8: Support MMC2 controller.

andreiw, 2020-03-01:
Today Pi 4 UEFI only supports the Arasan SDHCI controller being routed to the SD card. This is the same controller as used on Pi 2/3. It is also present on Pi 4 but usually relegated to SDIO (for WiFi).

We should support MMC2.

This can be done in two approaches.

Approach 1
=========
According to Jared from NetBSD, the Arasan driver should work just fine At least the NetBSD driver does. So, have ConfigDxe set a PCD for the right base depending on which controller is chosen.

Approach 2
==========
According to Jared from NetBSD, the Arasan driver should work just fine At least the NetBSD driver does. So, refactor Arasan to be a driver-model compliant driver, and add another "discovery" driver, which would enable instantiating Arasan twice for both EMMC2 and the Arasan SDHCI.

Approach 3
=========
Leverage the non-enumerated PCI device support and try using the "full" official SDMMC stack (e.g. SdMmcPciHcDxe). Might provide a path for migrating the Arasan support as well (although would involve figuring out how to add the notion of random quirks/hacks to the Intel SDHCI stack)

Googulator, 2020-03-01:
I have actually tried Approach 1 before (not with ConfigDxe, but hardcoding the base for MMC2) - it doesn't work. Likely the NetBSD driver is more of an SDHCI-with-quirks model, while ours relies on quirks/proprietary behavior to work at all.

andreiw, 2020-03-01:
You also would need to update the code that asks the VPU to enable power/clock...

andreiw, 2020-03-01:
But thanks for the report.

samerhaj, 2020-03-22:
regardless of the approach, we should aim at implementing EFI_SD_MMC_PASS_THRU_PROTOCOL as required by UEFI 2.8 specification (section 2.6.2 - item 29)

Approach #3 gets us there relatively easily, and is already supported in EDK2 using:
  MdeModulePkg/Bus/Pci/NonDiscoverablePciDeviceDxe/NonDiscoverablePciDeviceDxe.inf
  MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe.inf
  MdeModulePkg/Bus/Sd/EmmcDxe/EmmcDxe.inf

With approach #3, you can work around platform specific quirks/hacks in SdMmcPciHcDxe using EDKII_SD_MMC_OVERRIDE_PROTOCOL. See for example the approach in SynQuacer :

edk2-platforms\Silicon\Socionext\SynQuacer\Drivers\PlatformDxe\Emmc.c

andreiw, 2020-04-01:
I've tried Approach 1 as well, and did not get anywhere. Stepping back, we'll first expose MMC2 to the OS (ACPI), then have an option to differentiate firmware SD routing vs OS SD routing. That should let us validate that at least the OS can use the SD card via MMC (so we mux correctly and the clock is enabled/set)

jlinton, 2020-08-20:
Somewhat related, but the default sd routing, causes 

[ 3496.821065] mmc1: Timeout waiting for hardware cmd interrupt.
[ 3496.826828] mmc1: sdhci: ============ SDHCI REGISTER DUMP ===========
[ 3496.833265] mmc1: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[ 3496.839695] mmc1: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[ 3496.846126] mmc1: sdhci: Argument:  0x80000c08 | Trn mode: 0x00000000
[ 3496.852556] mmc1: sdhci: Present:   0x000f0001 | Host ctl: 0x00000001
[ 3496.858986] mmc1: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[ 3496.865416] mmc1: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[ 3496.871845] mmc1: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[ 3496.878275] mmc1: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[ 3496.884705] mmc1: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 3496.891135] mmc1: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 3496.897566] mmc1: sdhci: Cmd:       0x0000341a | Max curr: 0x00080008
[ 3496.903996] mmc1: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 3496.910426] mmc1: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 3496.916855] mmc1: sdhci: Host ctl2: 0x00000000
[ 3496.921288] mmc1: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 3496.927718] mmc1: sdhci: ============================================

to repeat endlessly when booted in DT mode on linux 5.7+

Changing the routing (via the BDS) to the mmc, fixes this, which in turn allows the wifi to appear. So when we get the driver in place, we should consider changing the default routing as well.

jlinton, 2021-01-21:
I'm closing this as the emmc2 changes are in edk2, the ACPI bits for linux/etc are not yet there but that #26

closed 2021-01-21 by jlinton