pftf/RPi4 #9: DWC2 controller driver (Type C) doesn't work with UFI devices (USB Floppy)

andreiw, 2020-03-01:
This a long known issue of the DWC2 driver... seen first in RaspberryPiPkg, then upstream RPi3 and now RPi4.

Not a very important bug, but a bug nonetheless, and there were unconfirmed reports of some USB1 devices (mass storage) and HID (keyboards) not working, which all could be related.

I need to get a USB2 analyzer such as https://shop.lambdaconcept.com/home/35-usb2-sniffer.html and have a look.

jlinton, 2020-03-31:
I've got a really old protocol analyzer (doesn't do any signal checking).
If you give me a bit more detail about what is failing and what you want captured, I can do it. 

Although, I've hacked on a variation of this driver for the hikey. In that case the dw controller wasn't doing split transactions correctly and there wasn't much that could be done about it, outside of slowing the whole device tree down to USB1 speeds.

andreiw, 2020-04-01:
Thanks Jeremy. The test is pretty simple - attach a USB floppy (UFI, this uses CBI).

I've yet to see any USB floppies work with the DWC2 driver, and I've tried a few (branded HP ones, Dell ones, no-name USB<->floppy adapters, etc).

Of interest may be the full DWC2 driver history: https://github.com/andreiw/RaspberryPiPkg/commits/master/Drivers/DwUsbHostDxe

E.g when I picked this driver up it had no periodic interrupt transfers nor split transfer support... needless to say, considering my massive USB experience (which is none at all), I may have implemented these in a buggy fashion. It's good enough for most (all?) keyboards and for some USB1 flash drives, but clearly something is not very good about it.

andreiw, 2020-04-14:
Two interesting data points with latest v1.7 release on Pi 4. I suggest testing with the HP drive or more other drivers. We /have/ gone from no drive ever working on dwc2 to now some kind of working...if my USB HID travails are any indication, it would mean there's a minor corner case somewhere in the controller code still missing.

With Dell drive I can successfully do I/O. With HP drive, it "hangs" for a bit, then shows a FS with no files (and can't create any), even though floppy has files.

My BusLink FDD1-based 5.25 enclosure works fine as well (in case you have a BusLink FDD1 3.5 drive

[image]

[image]
[image]

andreiw, 2020-04-14:
I guess it goes without comment that these all work fine on xHCI, including that HP one.

jlinton, 2020-05-08:
[rpi_floppy_disk_convert.zip](https://github.com/pftf/RPi4/files/4601518/rpi_floppy_disk_convert.zip)
[rpi_floppy_no_disk.zip](https://github.com/pftf/RPi4/files/4601519/rpi_floppy_no_disk.zip)

Here are a couple traces from the dw2 on the rpi4 in edk2 while booting to the bds.

One with a floppy in the drive, one without.

They can be viewed with.
"CATC 2500H, USBTracer/Trainer & Advisor Classic USB Protocol Suite"
from https://teledynelecroy.com/support/softwaredownload/documents.aspx?standardid=4

with just a free login. I think in general lecroy provides the viewer/analyzer software without to many restrictions because its licensed for capture purposes (aka buy the hardware/license to capture, but anyone can view the result).

andreiw, 2020-05-12:
Thanks Jeremy. Will take a look.