pftf/RPi4 #23: Upstream TF-A doesn't set up a stack

andreiw, 2020-03-06:
Usually BL31 never is the first stage for TF-A to run, consequently, the expectation is that BL2 will handle the stack setup. That's how the Pi 3 TF-A works..

Pi 4 TFA is a "bare" BL31. I don't see a plat/common/aarch64/platform_mp_stack.S getting linked in, and I don't see a plat_set_my_stack being done by ./bl31/aarch64/bl31_entrypoint.S.

So it is highly likely that:
- the stack pointer is a random value (e.g. 0 or similar)
- the stack pointer is shared by all CPUs 

This might explain the following observations:
- All ACS tests relying on PSCI fail
- I've seen CPU lock-ups in Linux relating to EA handling in TF-A. i.e. if I cause a repeatable stream of EAs, and maxcpus > 1, then it will lock up.

andreiw, 2020-03-10:
It's not a stack problem. The stack is actually set up inside el3_common_macros.S, inside _init_c_runtime macro.

andreiw, 2020-03-10:
The problem with the ACS tests is caused by PSCI_OFF. It's incomplete/incorrect on the Pi, putting the CPU in a wfi, where it should instead be parking it in the mailbox.

andreiw, 2020-03-10:
I kind of "fixed" it by writing to RMR_EL3 to simulate a reboot, but this will only work for secondary cores (which instantly go into the holding pen).

Sill need to see how to make sure PSCI_OFF for CPU 0 will work. This shouldn't be too bad, just need to special case plat_get_my_entrypoint to check the holding pen for BSP, and program the CPU0 holding pen with bl31_warm_entrypoint

andreiw, 2020-03-10:
ACS tests that use PSCI now pass

andreiw, 2020-03-26:
This is being worked in upstream TF-A now

andreiw, 2020-03-27:
This looks like the latest patch set - https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/3749/2

We're close now

pbatard, 2020-04-21:
With the update to TF-A v2.3, this should now be solved.

closed 2020-04-21 by pbatard