pftf/RPi4 #24: USB Input Devices don't function in Linux outside of UEFI, if DT mode is used

AdrianKoshka, 2020-03-09:
Problem: When I boot into Debian buster, on firmware v1.5, in DT mode, USB input devices cease to work. These are USB 2.0 input devices, attached to the USB 2.0 ports.

Repro instructions:

1. Burn [this iso (Debian buster arm64 xfce)](https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/debian-10.3.0-arm64-xfce-CD-1.iso) to a flash drive
2. put the UEFI into DT mode
3. Boot Debian
4. Select "install" at the bootloader
5. notice your input devices no longer work

pbatard, 2020-03-09:
Yup, DT mode isn't expected to work. Please remember that you are using __EXPERIMENTAL__ firmware, which basically means that you will find many broken features. And that is the reason why ACPI is enabled by default.

If you want stable firmware, where most of the features work, you will probably have to wait a few more months...

andreiw, 2020-03-09:
Thanks for the report. Will take a look at least as far as scoping out what this is.

andreiw, 2020-03-26:
I think someone on Discord mentioned that this is due to lack of upstream DT/PCIe support for Pi 4. Actually going to close this off as not a bug we're tracking (unless there's conclusive evidence that with DT and without UEFI) this is still a problem with 64-bit OSes.

closed 2020-03-26 by andreiw