pftf/RPi4 #26: Upstream Linux ACPI enablement for Arasan SDHCI

andreiw, 2020-03-27:
Local microSD card won't be available in Linux when booting in ACPI mode.

pbatard, 2020-03-27:
I guess there are 2 ways to go about this:
1. Update https://github.com/torvalds/linux/blob/master/drivers/mmc/host/sdhci-of-arasan.c to add ACPI bindings (similar to what we did for genet)
2. Update the generic https://github.com/torvalds/linux/blob/master/drivers/mmc/host/sdhci-acpi.c so that it works with Arasan.

I remember having a quick look at option 2 but not getting much out of it because it seemed to require additional quirks, therefore I suspect our best option is with 1. But I'm putting it out there just in case.

andreiw, 2020-03-27:
I think option 1 has a clearer path, esp. in light of all the quirks necessary for Arasan.

jlinton, 2020-04-21:
Hmmm, the first is chuck full of clock and of properties. The second (at least in the intel case) appears to make fairly heavy use _DSM() methods to configure the SD. This is potentially a nice case for actual platform abstraction if the speed detection and everything is done in AML. Reuse the GUID it might not even need much in the way of kernel code changes...

jlinton, 2020-04-21:
There is also this one: https://github.com/torvalds/linux/blob/master/drivers/mmc/host/sdhci-pci-arasan.c

andreiw, 2020-04-21:
The acpi one might be a good template to follow...that said, the “Arasan” controller IIRC is a bit weird - it’s SDHCI, yet only 32-bit accesses (and SDHCI really likes 1-byte registers, so everything is jammed together), and there are some domain crossing bugs. So sdhci-acpi probably won’t work “as-is”

macromorgan, 2020-08-14:
Near as I can tell it's not the sdhci-of-arasan that manages this device on the Pi, it's the sdhci-iproc driver that does (as evidenced by the links on the devicetree to brcm,bcm2835-sdhci and brcm,bcm2711-emmc2).  As fortune would have it this driver has both of and ACPI references in it.  For giggles, I added the following lines to this driver on my Debian 5.7.10 kernel and got the following in my dmesg:
`[ 3471.057265] sdhci_iproc: loading out-of-tree module taints kernel.
[ 3471.058397] mmc0: Hardware doesn't specify timeout clock frequency.
[ 3471.082512] mmc0: SDHCI controller on BCM2855:00 [BCM2855:00] using PIO`

It looks like the other SD driver in use is the mmc-bcm2835, but I don't see that it has any ACPI bindings.

macromorgan, 2020-08-14:
Some more information to help (I know almost nothing about C or ACPI Language aside from a little bit of work I've done on Coreboot).  This is from my active devicetree on a Pi 4 running 64 Bit Raspberry Pi OS:

`	aliases {`
`		mmc0 = "/emmc2bus/emmc2@7e340000";`
`		mmc1 = "/soc/mmcnr@7e300000";`
`};`
`        emmc2bus {`
`                dma-ranges = < 0x00 0xc0000000 0x00 0x00 0x40000000 >;`
`                #address-cells = < 0x02 >;`
`                #size-cells = < 0x01 >;`
`                compatible = "simple-bus";`
`                ranges = < 0x00 0x7e000000 0x00 0xfe000000 0x1800000 >;`
`                phandle = < 0xce >;`

`                emmc2@7e340000 {`
`                        vqmmc-supply = < 0x24 >;`
`                        interrupts = < 0x00 0x7e 0x04 >;`
`                        clocks = < 0x07 0x33 >;`
`                        broken-cd;`
`                        vmmc-supply = < 0x25 >;`
`                        compatible = "brcm,bcm2711-emmc2";`
`                        status = "okay";`
`                        reg = < 0x00 0x7e340000 0x100 >;`
`                        phandle = < 0x3a >;`
`                };`
`        };`

`                mmcnr@7e300000 {`
`                        pinctrl-names = "default";`
`                        pinctrl-0 = < 0x1f >;`
`                        bus-width = < 0x04 >;`
`                        non-removable;`
`                        interrupts = < 0x00 0x7e 0x04 >;`
`                        clocks = < 0x07 0x1c >;`
`                        dma-names = "rx-tx";`
`                        compatible = "brcm,bcm2835-mmc\0brcm,bcm2835-sdhci";`
`                        status = "okay";`
`                        reg = < 0x7e300000 0x100 >;`
`                        phandle = < 0x38 >;`
`                        dmas = < 0x0b 0x0b >;`
`                        brcm,overclock-50 = < 0x00 >;`
`                };`

jlinton, 2020-11-19:
[ 1297.817442] mmc0: SDHCI controller on ACPI [PNP0D40:00] using PIO
[ 1297.878750] mmc0: new SDHC card at address d81a
[ 1297.883998] mmcblk0: mmc0:d81a SU08G 7.40 GiB 
[ 1297.892139]  mmcblk0: p1

I've got it running with the sdhci_acpi driver, its not a large change, but I'm trying to figure out how to do it without adding a custom _HID or rpi specific changes. At the moment its really just feeding the right voltage & clock, capabilities and assuring that SDHCI_QUIRK_DATA_TIMEOUT_USES_SDCLK gets set.

jlinton, 2020-12-09:
I've got edk2 accessing the SD with either the arasan or the emmc2, its also working in linux at the moment, although I'm using the PNP0D40 as above for _both_ controllers because it doesn't require adding a custom ACPI HID to the sdhci_iproc driver (and the linux patch to quirk the caps registers has been merged). The problem with this, is that there is a comment in the iproc driver about a clock domain crossing problem on !data registers. So while it appears to be working with my minimal testing, i'm concerned we may be running without proper bug workarounds.

(the wifi isn't yet functional, that is what i'm working on now, along with some general cleanups to the patches)

jlinton, 2020-12-09:
Also: [root@rpi4-f33 brcm]# iw dev
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr 22:e9:86:96:e5:6d
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 5
                wdev 0x1
                addr f6:69:14:ea:8e:7d
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm

So the wifi, is sorta working, which is how it behaves in DT on my machine too (sorta working).

jlinton, 2020-12-15:
I've posted just the edk2 driver updates to allow switching between the arasan and emmc2.

The ACPI table updates, and possible linux sdhci_iproc binding just needs a HID at this point.

There is also the somewhat related issue, that the kernel firmware needs to have the smbios machine name linked for the wifi firmware.

aiden1989acw, 2020-12-23:
> I've posted just the edk2 driver updates to allow switching between the arasan and emmc2. [...]

Is there a Driver port for BCM2855 for Windows on Arm 64Bit available? How much work is required to port over? as this would greatly assist the guys over at WoR with adding Wifi functionality.

I believe we need to port a driver over to free up the Arasan Driver for SDIO 3.0, this currently tied up with the SD card.

The usage of the BCM2855 would greatly increase SD card performance too rather than using the Arasan BCM2847, i'm led to believe.

Googulator, 2021-01-01:
In theory, it should work with the inbox sdbus driver, if the ACPI ID and the correct Registry parameters qre added to the inf. In practice, I'm still trying to figure out the correct parametrization.

aiden1989acw, 2021-01-05:
> 
> 
> In theory, it should work with the inbox sdbus driver, if the ACPI ID and the correct Registry parameters qre added to the inf. In practice, I'm still trying to figure out the correct parametrization.

That should work, currently having issues with the latest RPI_EFI.fd to implement the emmc2 driver, it will not load the uefi, any assistance would be greatly appreciated, having an asynchronous error on boot

jlinton, 2021-01-11:
Can you try building the latest edk2-platforms against just the base pi4 PFTF image? There are problems with the latest versions of the rpi foundation provided firmware images so you need to stick to the ones being shipped here. 

You won't need to merge any patches as the base emmc2 driver should now be in edk2-platforms. 

The ACPI tables, and linux kernel enablement patches, and wifi firmware patches are under review on the appropriate mailing lists, but I would expect them to land sometime this week in their respective repos (in the case of linux in -next).

Googulator, 2021-01-11:
To update my previous comment for emmc2 in Windows: it works if you disable DMA (DmaMethod=1 - likely 1GB DMA limit issue) and UHS (DisableUhsSupport=1) in Registry. Of course, this limits it to 25MB/s with high CPU overhead.

jlinton, 2021-01-13:
> 
> 
> To update my previous comment for emmc2 in Windows: it works if you disable DMA (DmaMethod=1 - likely 1GB DMA limit issue) and UHS (DisableUhsSupport=1) in Registry. Of course, this limits it to 25MB/s with high CPU overhead.

Can you remove the _DSD entry from the emmc2 block that disables ADMA2 and see if that helps with the DMA?

jlinton, 2021-01-21:
https://git.kernel.org/pub/scm/linux/kernel/git/ulfh/mmc.git/commit/?h=next&id=22510bcd5547f6847c821c12785b4ed435c61662

SquallATF, 2021-01-21:
@jlinton 
Arasan SDHCI route to sd or wifi bouth can work, but emmc2 route to sd wan't work.
sd card boot faile with `mmc0: error -110 whilst initialising SD card`
if switch ACPI to DT sd card can boot.

- ACPI boot log

```
[    1.904824] sdhci: Secure Digital Host Controller Interface driver
[    1.910996] sdhci: Copyright(c) Pierre Ossman
[    1.915373] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
[    1.922504] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.928431] mmc0 bounce up to 128 segments into one, max segment size 65536 bytes
[    1.928489] NET: Registered protocol family 10
[    1.940754] Segment Routing with IPv6
[    1.944448] NET: Registered protocol family 17
[    1.948900] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    1.961846] 8021q: 802.1Q VLAN Support v1.8
[    1.966157] Key type ._fscrypt registered
[    1.966969] mmc1: SDHCI controller on BCM2847:00 [BCM2847:00] using PIO
[    1.970163] Key type .fscrypt registered
[    1.976828] mmc0: SDHCI controller on BRCME88C:00 [BRCME88C:00] using DMA
[    1.980696] Key type fscrypt-provisioning registered
[    1.992615] of_cfs_init
[    1.995117] of_cfs_init: OK
[    1.998077] Waiting for root device /dev/mmcblk0p2...
[    2.016153] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.023254] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.030357] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.038750] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.045852] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.092186] mmc0: error -110 whilst initialising SD card
[    2.111693] random: fast init done
[    2.153138] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    2.156934] mmc1: new SDIO card at address 0001
[    2.307738] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    2.315911] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.323038] usb 1-1: Product: USB2.0 Hub
[    2.327896] hub 1-1:1.0: USB hub found
[    2.331830] hub 1-1:1.0: 4 ports detected
```
- DT boot log
```
[    0.258813] sdhci: Secure Digital Host Controller Interface driver
[    0.258821] sdhci: Copyright(c) Pierre Ossman
[    0.259011] mmc-bcm2835 fe300000.mmcnr: could not get clk, deferring probe
[    0.259152] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
[    0.259159] sdhci-pltfm: SDHCI platform and OF driver helper
[    0.262882] NET: Registered protocol family 10
[    0.263317] Segment Routing with IPv6
[    0.263353] NET: Registered protocol family 17
[    0.263373] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    0.263381] 8021q: 802.1Q VLAN Support v1.8
[    0.263498] Key type ._fscrypt registered
[    0.263503] Key type .fscrypt registered
[    0.263508] Key type fscrypt-provisioning registered
[    0.266044] uart-pl011 fe201000.serial: there is not valid maps for state default
[    0.266106] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    0.266158] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 21, base_baud = 0) is a PL011 rev2
[    1.443873] printk: console [ttyAMA0] enabled
[    1.451981] bcm2835-aux-uart fe215040.serial: there is not valid maps for state default
[    1.460103] bcm2835-aux-uart fe215040.serial: error -ENOSPC: unable to register 8250 port
[    1.468295] bcm2835-aux-uart: probe of fe215040.serial failed with error -28
[    1.475445] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    1.482893] mmc-bcm2835 fe300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    1.488899] mmc-bcm2835 fe300000.mmcnr: DMA channel allocated
[    1.520832] of_cfs_init
[    1.523323] of_cfs_init: OK
[    1.546607] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    1.553571] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.560550] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.561819] mmc0: SDHCI controller on fe340000.emmc2 [fe340000.emmc2] using ADMA
[    1.568693] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    1.578951] Waiting for root device /dev/mmcblk0p2...
[    1.578988] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.591808] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.625201] random: fast init done
[    1.653644] mmc1: new high speed SDIO card at address 0001
[    1.702161] mmc0: new ultra high speed DDR50 SDXC card at address aaaa
[    1.709051] mmcblk0: mmc0:aaaa SC64G 59.5 GiB
[    1.718302]  mmcblk0: p1 p2
[    1.736436] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
```

jlinton, 2021-01-21:
That is probably because you merged my last public patch. You should update the caps_mask:

-      Package () { "sdhci-caps-mask", 0x0000000000080000 },
+      Package () { "sdhci-caps-mask", 0x0000000504080000 },^M

That fully disables the 1.8V switch required for newer SD cards, which isn't supported by the standard SDHCI regs on the rpi, and since we are ACPI you can't use a DT voltage reg/mailbox to do it either. I discovered that a couple weeks back when running more tests (although I'm not happy with the _DMA in the original set yet either).

jlinton, 2021-01-21:
That change also brings ~50MB/sec SD reads with DT down to ~25MB/sec when I was testing it.

jlinton, 2021-02-01:
Hopefully final patch set posted here:

https://edk2.groups.io/g/devel/topic/patch_v2_0_4_rpi_sd_wifi/80300530?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80300530

That version enables/disables the emmc ACPI/SSDT table when the Arasan is selected, as well as adding a DMA/PIO enable/disable menu item. Its also defaulting the emmc2 to PIO mode too. That is the big hammer way to avoid all the _DMA issues in linux.

jlinton, 2021-02-22:
Since Ard merged https://github.com/tianocore/edk2-platforms/commit/311ed3c79047e2a7ad001ab94b240ac6558bf06c, I'm going to close this with a reference to various improvements that can be made in https://github.com/pftf/RPi4/issues/134

closed 2021-02-22 by jlinton