pftf/RPi4 #27: Updated USB networking drivers

samerhaj, 2020-03-30:
We should pickup the new ASIX USB NIC drivers (currently in [this branch](https://github.com/samerhaj/edk2-platforms/commits/ASIX_USB_Networking)) once up-streamed to EDK2:

- [Ax88772c](https://github.com/samerhaj/edk2-platforms/tree/ASIX_USB_Networking/Drivers/OptionRomPkg/Bus/Usb/UsbNetworking/Ax88772c) (version 2.8.0 replacing Ax88772c ver 1.0)
- [Ax88179](https://github.com/samerhaj/edk2-platforms/tree/ASIX_USB_Networking/Drivers/OptionRomPkg/Bus/Usb/UsbNetworking/Ax88179)

In addition, we should look at picking up some of the compatible adapters with different VendorId/ProductID (for example as seen in the [Linux ASIX driver](https://github.com/torvalds/linux/blob/7111951b8d4973bda27ff663f2cf18b663d15b48/drivers/net/usb/asix_devices.c#L121))

rgl, 2020-03-30:
I was able to successfully test the Ax88179 drivers with the instructions that I've placed at https://github.com/rgl/raspberrypi-uefi-edk2-vagrant/compare/feature-switch-submodules-howto, which I kindly ask if you can review them? :-)

samerhaj, 2020-05-12:
Not needed anymore, since we have the GENET driver. Closing

closed 2020-05-12 by samerhaj