pftf/RPi4 #28: RPi 4 stuck during UEFI init

sassmann, 2020-04-01:
About 20% of the time when powering up the RPi4 it gets stuck indefinitely at either
```
NOTICE:  BL31: v2.2(release):v2.1-997-gc293471b
NOTICE:  BL31: Built : 17:12:49, Mar  5 2020
UEFI firmware (version UEFI Firmware  built at 03:22:40 on Mar 31 2020)
```
or shortly after at
`ESC (setup), F1 (shell), ENTER (boot)...`

Tested with a 2GB RPi 4 and fw 1.5 as well as 1.7. Anybody else seeing this?

pbatard, 2020-04-01:
Since you have serial, can you try with the DEBUG version of the firmware from https://ci.appveyor.com/project/pbatard/RPi4/build/artifacts and let us know where exactly it seems to freeze?

I don't remember experiencing a freezout after the `ESC / F1 / Enter` options are being displayed, but I do sometime seems to experience __temporary__ delays during USB 3.0 init, which I always assumed have to do with my using of a "smart" USB switch for keyboard and mouse, that doesn't always seem to properly connect devices (since I'm also observing reluctant connections from within OSes that don't use UEFI at all). After a while, the firmware always seems to boot however...

sassmann, 2020-04-01:
Here's the log from your DEBUG build:
```
Loading driver 100C2CFA-B586-4198-9B4C-1683D195B1DA
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B 35F63D40
add-symbol-file /home/appveyor/projects/rpi4/Build/RPi4/DEBUG_GCC5/AARCH64/Platform/RaspberryPi/Drivers/ArasanMmcHostDxe/ArasanMmcHostDxe/DEBUG/ArasanMMCHost.dll 0x33AA7000
Loading driver at 0x00033AA6000 EntryPoint=0x00033AA98EC ArasanMMCHost.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 35F63C98
ProtectUefiImageCommon - 0x35F63D40
  - 0x0000000033AA6000 - 0x0000000000008000
SetUefiImageMemoryAttributes - 0x0000000033AA6000 - 0x0000000000001000 (0x0000000000004008)
SetUefiImageMemoryAttributes - 0x0000000033AA7000 - 0x0000000000005000 (0x0000000000020008)
SetUefiImageMemoryAttributes - 0x0000000033AAC000 - 0x0000000000002000 (0x0000000000004008)
InstallProtocolInterface: 3E591C00-9E4A-11DF-9244-0002A5F5F51B 33AAC030
InstallProtocolInterface: 964E5B21-6459-11D2-8E39-00A0C969723B 35F639C8
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 35F63798
MMCSendCommand(371): MMC_CMD1 ERRI MmcStatus 0x18040
Card is SD2.0 => Supports high capacity
Response: 0x1CE
        - READY_FOR_DATA
        - State: Idle
Card should be SD
Card should be SD
Card should be SD
Card should be SD
Card should be SD
Card should be SD
High capacity card.
- PrintOCR Ocr (0xC0FF8000)
        - Card operating voltage: 2.7 to 3.6
        - AccessMode: Block Mode (0x2)
        - PowerUp
- PrintCID
        - Manufacturing date: 10/8
        - Product serial number: 0x701CE89F
        - Product revision: 128
        - OEM ID: DS
- PrintRCA: 0x8FE40520
        - Status: 0x520
        - RCA: 0x8FE4
- PrintCSD Version 2.00/High Capacity
        - Supported card command class: 0x5B5
        - Max Speed: 3.0 * 10Mbit/s
        - Maximum Read Data Block: 512
        - Maximum Write Data Block: 512
        - Format (0): Hard disk-like file system with partition table
Found SD Card for Spec Version 2.0
Controller doesn't support speed change
Controller doesn't support bus width change
InstallProtocolInterface: CE345171-BA0B-11D2-8E4F-00A0C969723B 35F62BA0
 BlockSize : 512
 LastBlock : ECDFFF
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 35F62B18
InstallProtocolInterface: 964E5B21-6459-11D2-8E39-00A0C969723B 35F59030
InstallProtocolInterface: 8CF2F62C-BC9B-4821-808D-EC9EC421A1A0 35F590E8
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 35F59E98
InstallProtocolInterface: 964E5B21-6459-11D2-8E39-00A0C969723B 35F592B0
InstallProtocolInterface: 8CF2F62C-BC9B-4821-808D-EC9EC421A1A0 35F59368
InstallProtocolInterface: CE345171-BA0B-11D2-8E4F-00A0C969723B 35F59820
 BlockSize : 512
 LastBlock : 7FFFF
InstallProtocolInterface: 964E5B22-6459-11D2-8E39-00A0C969723B 35F56030
Installed Fat filesystem on 35F62A18
FSOpen: Open 'RPI_EFI.FD' Success
FSOpen: Open 'RPI_EFI.FD' Success
LibGetTime: RtcEpochSeconds non volatile variable was not found - Using compilation time epoch.
LibGetTime: RtcEpochSeconds non volatile variable was not found - Using compilation time epoch.
LibGetTime: RtcEpochSeconds non volatile variable was not found - Using compilation time epoch.
LibGetTime: RtcEpochSeconds non volatile variable was not found - Using compilation time epoch.
ValidateWrittenBlockCount(90): expected 128 != gotten 127
MMCSendCommand(329): not ready for MMC_CMD13 PresState 0x1FFF0506 MmcStatus 0x40
MMCSendCommand(329): not ready for MMC_CMD13 PresState 0x1FFF0506 MmcStatus 0x40
MMCSendCommand(329): not ready for MMC_CMD13 PresState 0x1FFF0506 MmcStatus 0x40
MMCSendCommand(329): not ready for MMC_CMD13 PresState 0x1FFF0506 MmcStatus 0x40

```
The last message repeats over and over again. Looks like an issue with the microSD card.

pbatard, 2020-04-01:
> Looks like an issue with the microSD card.

There you go. Please use a different SD card, and feel free to report if it still doesn't work.

I will close this issue now.

closed 2020-04-01 by pbatard

sassmann, 2020-04-01:
Agreed. Thank you Pete.