pftf/RPi4 #29: Rename RPI_EFI to RPI4_EFI

midyukov-anton, 2020-04-06:
For ease of testing on RPI3 and RPI4, I host both RPI3_EFI.fd and RPI4_EFI.fd. But there is a problem, the settings are not saved, as they try to write to RPI_EFI. I did a workaround. When assembling rpm in the spec, I change RPI_EFI to RPI3_EFI and RPI4_EFI.

Can you set this to a variable?

I make the changes in three files:
edk2-platforms/Platform/RaspberryPi/Drivers/VarBlockServiceDxe/VarBlockService.c
edk2-platforms/Platform/RaspberryPi/RPi3/RPi3.fdf
edk2-platforms/Platform/RaspberryPi/RPi4/RPi4.fdf

I use such config.txt:
```
arm_64bit=1
disable_commandline_tags=2
disable_overscan=1
enable_uart=1
uart_2ndstage=1

[pi3]
armstub=RPI3_EFI.fd

[pi4]
armstub=RPI4_EFI.fd
device_tree_address=0x1f0000
device_tree_end=0x200000
device_tree=bcm2711-rpi-4-b.dtb
dtoverlay=miniuart-bt

[all]
```

Link to my spec:
http://git.altlinux.org/gears/e/edk2-rpi.git?p=edk2-rpi.git;a=blob;f=edk2-rpi.spec;h=5517e60e2c0a3dc187baa23243b42941d15125ae;hb=e2622bee1c81ea2baa9b4a61d9b84d8d12ad94f6

samerhaj, 2020-04-06:
long term, there is a plan to write the variables to SPI flash: https://github.com/pftf/RPi4/issues/6

For renaming the files, I would imagine moving the variables to their own file (in a new FD+FV) is a more sensible approach. This way, both RPi3 and RPI4 could read/write variables from say RPI_EFIVAR.fd. This also helps in the issue with losing settings when updating the RPI_EFI.fd file

midyukov-anton, 2020-04-06:
> For renaming the files, I would imagine moving the variables to their own file (in a new FD+FV) is a more sensible approach. This way, both RPi3 and RPI4 could read/write variables from say RPI_EFIVAR.fd. This also helps in the issue with losing settings when updating the RPI_EFI.fd file

I'm talking about something else. About the possibility to do one assembly for RPi3 and RPi4 right now. It is convenient and time saving.

Writing variables to a separate file is a great idea. But I'm talking about something else now :-)

andreiw, 2020-04-06:
Same RPI_EFI.FD for both 3 and 4 is not possible atm (and wouldn’t want to share vars).

I like the idea of making 3- and 4- specific FD names. That would solve the immediate problem of using same SD card on both Pies.

Marcinoo97, 2020-04-09:
> For ease of testing on RPI3 and RPI4, I host both RPI3_EFI.fd and RPI4_EFI.fd. But there is a problem, the settings are not saved, as they try to write to RPI_EFI. I did a workaround. When [...]

I did the same thing as you over a month ago.

jlinton, 2021-12-09:
I think sorta the point with the rpi4 and SPI variables, is that you can rename the rpi4 image to anything you want and it should work. It will then only be the rpi3 that needs a firmware image named RPI_EFI.fd

jonesmz, 2023-08-10:
I work on a project to create a Gentoo linux derived OS image for the raspberry pi 3, 4, and 0w2, and I want to use the UEFI firmware from this project.

However, without out-of-the-box support for running a single UEFI install on each of the three boards, I'm kind of stuck.

Can this change be made in the future?

Something i can make a pull request for without a lot of research and head scratching?