pftf/RPi4 #33: Raspberry Pi 4UEFI firmware does not recognize 32-bit boot program

bzy-080408, 2020-04-14:
Raspberry Pi 4UEFI firmware does not recognize 32-bit boot program, in the boot manager to choose no use, bar config.txt file changed to arm_64bit=0 or delete this line to display a rainbow screen

andreiw, 2020-04-14:
This is a 64-bit UEFI implementation, and consequently it only supports 64-bit OS loaders and operating systems.

closed 2020-04-14 by andreiw

bzy-080408, 2020-04-14:
Do you have a 32-bit UEFI firmware?

pbatard, 2020-04-14:
If you think you __need__ a 32-bit firmware, you can:
* Clone [edk2-platforms](https://github.com/tianocore/edk2-platforms/), which is the Open Source project the firmware releases you see on this site are built from
* Alter the code to add 32-bit support
* Submit your 32-bit support proposal to the edk2-devel mailing list so that it gets reviewed and integrated. If you do that be prepared to be able to justify why 32-bit support is a necessity for the firmware.

Then, once it has been reviewed and integrated, we'll be happy to publish 32-bit firmware images here.

But you have to understand that none of the current developers see any interest for having a 32-bit firmware (and you certainly haven't provided us with any clue as to why you think this is _must have_), so we aren't going to devote our time, which is in short supply, to add a feature that we currently don't think will bring anything to Raspberry Pi users. Pi users already have a plethora of 64-bit OSes they can use, and therefore have little "need" for 32-bit support. We tend to only add features that we see as benefiting Raspberry Pi end-user community as a whole, and short of an explanation that clarifies why your request would benefit many many users, we certainly do not see the ability to produce both an ARM and a AARCH64 firmware for the Pi as something we want to invest any time on.

bzy-080408, 2020-04-15:
Could you tell me how to build in a windows environment, I'm not very good at programming.

pbatard, 2020-04-15:
I'm afraid not. Don't take it the wrong way but we do expect people interested in recompiling the firmware to be resourceful enough to figure out how to do it on their own, as none of the people here have much time to perform hand-holding activities. If you are interested in recompiling the firmware, I would also strongly advise you to use Linux (possibly in a VM) rather than Windows.