pftf/RPi4 #35: Multi-language support

bzy-080408, 2020-04-15:
When to walk multilingual support

pbatard, 2020-04-15:
Please rephrase your question, as it is not clear what you are asking.

Are you asking about having more than English in the UEFI settings menu? Or is it something else.

And you should also realize that any question that starts with _"When will feature X be available?"_ in a firmware that is clearly flagged as __EXPERIMENTAL__ will be answered with _"We don't know"_.

bzy-080408, 2020-04-15:
I mean adding Chinese, German, Japanese, etc.

pbatard, 2020-04-15:
Answer: When someone sends a patch to the edk2-devel mailing list to add support for them.

I would however __wait__ until the firmware is no longer flagged as EXPERIMENTAL, since the menus are not finalized yet (and neither is the firmware), so it makes very little sense to add support for additional languages now.

Please, do understand that this is __NOT__ a finalized firmware. It is __NOT__ ready for anything besides testing, so please don't ask for features that only make sense for a finalized firmware.

I will therefore close this issue now.

bzy-080408, 2020-04-15:
OK,but what's the difference between start english and english?

bzy-080408, 2020-04-15:
It doesn't seem to make any difference.

pbatard, 2020-04-15:
Sorry, but I don't understand your question at all. What is "start english"? Where do you see that?

bzy-080408, 2020-04-15:
First option

pbatard, 2020-04-15:
That comes from edk2. Again, we are not planning to do anything that relates to the language until the firmware is finalized. That also applies to answering questions like "Why is menu X this way and not that way?". This is not a productive use of our time. Thank you.

samerhaj, 2020-04-15:
I think this is in reference to these current language options , which seem a bit confusing:

1. what is the difference between English/French and Standard English/Standard French?
2. Should we remove French/Standard French advertised support since none of the strings in the HII drivers are actually translated . Seems like just a couple of PCDs to set in the DSC file:

[image]

pbatard, 2020-04-15:
I think there are other priorities right now, but if you want to go for it, please go ahead.

samerhaj, 2020-06-19:
Will remove French

closed 2020-08-13 by ardbiesheuvel