pftf/RPi4 #36: Limiting memory

bzy-080408, 2020-04-16:
Change the limit memory to 3072MB to your own input value or increase the option to limit the limit to 2048MB and 1024MB.
Using translation software, possible translation is incorrect

pbatard, 2020-04-16:
Can you indicate why you think you need this feature? What exactly are you trying to accomplish where the 3072 MB limit won't work?

I appreciate that you English is not your first language, but if you request a feature, you should provide a bit more information about what exactly you are trying to do, that warrants the need for the feature. Unless we can estimate how much it is going to be useful for __other users__, we're unlikely to add every feature that is being requested. So you need to start elaborating a bit more about what it is that you are trying to accomplish. What is your goal?

bzy-080408, 2020-04-16:
For users using Windows 10arm, the memory limit is 1024MB, otherwise the USB controller will not function properly. In fact, you can do this by commanding BCD files, but the process is cumbersome.

pbatard, 2020-04-16:
As long as we don't have xHCI support, I don't think we want to take any action when it comes to limiting memory for Windows in the firmware, as the expectation is that, once we are able to run Windows with support for "backplate" USB devices, very few people will need the 1 GB limit and the BCD workaround should be enough.

In other words, I think it is way too early in this experimental firmware, and with Windows support far from being finalized, to take any action with regards to assuming that we will need a 1 GB limit for Windows in the long run.

But again, if you or anyone else thinks this is needed, you are free to submit a patch to edk2-devel.

bzy-080408, 2020-04-16:
I think you can change the option of limiting memory to 3072MB instead of setting it yourself.

pbatard, 2020-04-16:
I think you mean the opposite, and yes, we can add the change you want, but you have to understand that any change needs to be submitted to the EDK2 and go for review, since, as opposed to other projects, the code that is used to generate the firmware builds does not come from this project, but is held in the edk2, in https://github.com/tianocore/edk2, https://github.com/tianocore/edk2-platforms/ and https://github.com/tianocore/edk2-non-osi/.

So, adding your request means creating a formal patch submission to edk2, justifying why this is needed, getting it reviewed and getting is integrated, which, even for a simple change, is a rather time consuming process.

As such, we want to do that for changes that we feel are going to be needed in the long run. And as I explained, because we don't know yet how the Windows situation with the 1 GB limit is going to evolve (the expectation is that Windows will be able to use the full 4 GB once it can run with xHCI devices), it's hard to want to go through the above for this change.

But again, that's just my take on this, and, since this is Open Source, anybody who believes we should add an arbitrary memory limit rather than the 4 GB vs 3 GB toggle, can go through the process if they want to. I'm just saying that, personally, I don't currently see that change as a necessity.

andreiw, 2020-04-16:
You can use bcdedit to limit windows itself. There's no need to add an additional 1GB limit to UEFI as there is an OS workaround, and this workaround would only be necessary for Windows. Once we have xHCI working the limit won't matter.

closed 2020-04-16 by andreiw