pftf/RPi4 #37: Improve Arasan driver to fail gracefully when no card is present

andreiw, 2020-04-16:
Booting UEFI from USB or network still requires an SD card, because the Arasan driver will loop forever on I/O errors. Unfortunately the Pi Foundation didn't wire up the card detect pin so we have to do something more creative.

ehem, 2020-07-13:
With USB boot available on all variants of Raspberry PI, this now becomes a show-stopping issue.

Might getting the MMC2 controller working solve this issue?

Does the lower level loader provide any indications what method booted the Raspberry PI? (being loaded from a USB device is an implicit instruction to load from USB device)  Might it be possible to disable the Arasan driver via command-line?

pbatard, 2020-07-13:
@ehem: A patch was submitted more than a week ago with a fix for this. We are (more or less) waiting for it to be upstreamed at this stage. __Please please please__, do not assume that you get the full picture of what is really happening from the presence or absence of updates in this issue tracker. This is __NOT__ twitter: you won't find real-time updates of the work being done here. So if you really want to know what's going on, susbcribe to edk2-devel and monitor Pi related patches there, thank you.

commit 0c9ca2ec (2020-07-14): Update to EDK2 latest

* Closes #37
closed 2020-07-14 by pbatard