pftf/RPi4 #38: ASSERT [tftpDynamicCommand] /home/vagrant/rpi4-uefi/Build/RPi4/DEBUG_GCC5/AARCH64/ShellPkg/DynamicCommand/TftpDynamicCommand/TftpDynamicCommand/DEBUG/AutoGen.c(520): !EFI_ERROR (Status)

andreiw, 2020-04-16:
Also tracked on Pi 3 as https://github.com/pftf/RPi3/issues/11

andreiw, 2020-04-16:
`Here's what's around line 520 in AutoGen.c:

VOID
EFIAPI
ProcessLibraryConstructorList (
  IN EFI_HANDLE        ImageHandle,
  IN EFI_SYSTEM_TABLE  *SystemTable
  )
{
  EFI_STATUS  Status;

  Status = BaseDebugLibSerialPortConstructor ();
  ASSERT_RETURN_ERROR (Status);

  Status = UefiBootServicesTableLibConstructor (ImageHandle, SystemTable);
  ASSERT_EFI_ERROR (Status);

  Status = UefiRuntimeServicesTableLibConstructor (ImageHandle, SystemTable);
  ASSERT_EFI_ERROR (Status);

  Status = UefiHiiServicesLibConstructor (ImageHandle, SystemTable);
  ASSERT_EFI_ERROR (Status);

  Status = UefiLibConstructor (ImageHandle, SystemTable);
  ASSERT_EFI_ERROR (Status);

  Status = ShellLibConstructor (ImageHandle, SystemTable);
  ASSERT_EFI_ERROR (Status); // line 520

}

So obviously the result from the call to ShellLibConstructor() was unexpected.`

pbatard, 2020-04-21:
Should be fixed with latest build.

closed 2020-04-21 by pbatard