pftf/RPi4 #40: pipxe - No more network devices (is support for "GENET" in v1.9?)

blitmap, 2020-04-28:
Hello -

I'm doing something weird and I'm hoping I'm not offending anyone by talking about another project on this issue tracker.

This issue is more of a "Can you confirm this should not work yet?"  I believe the trouble is a lack of support for the NIC (GENET) ?

I've been following "pipxe":  https://github.com/ipxe/pipxe

Posted on their issue tracker was a lack of support for the RPi4.  A dev from this project replied saying the underlying UEFI firmware would need to support the NIC on the newer RPi4:

https://github.com/ipxe/pipxe/issues/2#issuecomment-604840301

This is what I see with the "image I've created":
```
  iPXE initialising devices...ok

  iPXE 1.0.0+ (3fe68) - Open Source Network Boot Firmware -- http://ipxe.org
  Features: DNS HTTP iSCSI TFTP AoE EFI Menu

  No more network devices

  iPXE initialising devices...ok

  iPXE 1.0.0+ (3fe68) - Open Source Network Boot Firmware -- http://ipxe.org
  Features: DNS HTTP iSCSI TFTP AoE EFI Menu

  No more network devices

  BdsDxe: No bootable option or device was found.
  BdsDxe: Press any key to enter the Boot Manager Menu.
```
At this point I do not have any control with a keyboard.  I believe at the end there it fell back to the UEFI bootloader?

My steps:

1) Copy the pipxe release here to your bootable FAT partition:
    https://github.com/ipxe/pipxe/releases/download/v20191103/sdcard.zip
2) Unzip the files from this to the same space, ovewriting maybe 5-6 files:
    https://github.com/pftf/RPi4/releases/download/v1.9/RPi4_UEFI_Firmware_v1.9.zip

(I used the default UEFI settings)

andreiw, 2020-04-28:
Hi!

So 1.9 doesn't have the GENET driver in there yet (that's for the NIC), but rest assured that work is ongoing to upstream the driver to make it part of the same release.

Are you also saying your USB keyboard doesn't work? Does it work before iPXE starts? I.e. just keep mashing ESC after power on Pi.

blitmap, 2020-04-28:
The USB keyboard does work during the UEFI portion.  I’ve gone in to
explore settings (I don’t yet know what I should do with the devicetree and
3gb limit).

It is only during pipxe that I can’t use the keyboard and I think that’s
because it’s panicking with no network devices

andreiw, 2020-04-29:
yeah it might be doing something interesting like getting all the USB devices to disconnect from their drivers... who knows

So, if you don't mind, I'll close this out. Lack of GENET in v19 itself is not a bug and it's already getting worked on :)

If you're willing to get your hands dirty you can try building https://github.com/pftf/edk2-platforms/tree/pi4_genet_samer (the wip branch for GENET work), although the standard disclaimers apply: you're on your own, we can't support you with building steps etc. However you're welcome to hang out on our Discord (https://discordapp.com/invite/fqRhc8y) channel #rpi4-uefi-dev

closed 2020-04-29 by andreiw

blitmap, 2020-04-29:
Absolutely appropriate to close this (not a bug) :)