pftf/RPi4 #41: DisplayDxe issues uncovered by SCT in v2.4 Arm Enterprise ACS

andreiw, 2020-04-30:
[image]
[image]
[image]

samerhaj, 2020-06-18:
A similar issue happens with SCT 2.4 on REL1.15:

https://raw.githubusercontent.com/pftf/acs-reports/master/v1.15/acs-enterprise-v2.4/SCT_RUN/Overall/Summary.log

```
------------------------------------------------------------
Arm ACS Version: v2.4
BltVideoBltBuffer_Func 
Revision 0x00010000
Test Entry Point GUID: B504931B-BC55-4E84-8229-F831623785FB
Test Support Library GUIDs: 
  1F9C2AE7-F147-4D19-A5E8-255AD005EB3E
------------------------------------------------------------
UEFI 2.6
Test Configuration #0
------------------------------------------------------------
Efi GraphicsOutput Protocol BltVideoBltBuffer Function Test - Blt VideotoBltBuffer Operation Test
------------------------------------------------------------
Logfile: "\SCT_RUN\Overall\Summary.log"
Test Started: 06/12/20  04:29p
------------------------------------------------------------

Current Device: VenHw(C5DEAE31-FAD2-4030-841B-CFC9644D2C5B)
------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
System hangs or stops abnormally. -- FAILURE
DE687A18-0BBD-4396-8509-498FF23234F1
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestInfrastructure/SCT/Framework/Execute/Execute.c:2073

Returned Status Code: Unsupported

BltVideoBltBuffer_Func : [FAILED]
  Passes........... 0
  Warnings......... 0
  Errors........... 1
```

samerhaj, 2020-11-16:
Duplicate of #97. Re-testing using v.1.21 FW

samerhaj, 2021-07-16:
Re-opning

This issue still occurs on ACS 3.0 and 1.28 firmware, running on RPi 400

- When running in automated/unattended ACS mode, I see the Synchronous Exception reported in #41
- When running in manual SCT mode (from the same ACS 3.0, same 1.28 FW, on the same machine, enter "sct -u" and select the Graphics Console Tests and run), the test passes