pftf/RPi4 #42: Arm Enterprise ACS 2.3 and 2.4 don't ever finish

andreiw, 2020-04-30:
[image]
[image]

It's actually supposed to look something like:
[image]

Note the **_Results were saved in your boot media. You may now turn off the system and remove it._** message.

andreiw, 2020-05-12:
I haven't tested this post the ACPI changes in v1.11 yet, but the expectation is that it should affect that and newer release too.

samerhaj, 2020-06-18:
I ran ACS 2.4 with REL1.15 and it finished completely.

There was a couple of places where SCT ran a test to shutdown/restart, but the RPi just shutdown, and I had to manually start it. Other than it, it completed all tests. Logs uploaded here: https://github.com/pftf/acs-reports/tree/master/v1.15/acs-enterprise-v2.4

should this issue be closed?