pftf/RPi4 #51: DisplayDxe: keeping virtual resolutions on by default gives fuzzy video in Linux

andreiw, 2020-05-08:
When booting via Frontpage (which sets 800x600), Linux will not change resolution and things will look bad on screens that can do better. The principle of least surprise would imply that folks booting Linux with UEFI would not get stuck at 800x600.

One solution is just to disable the virtual resolutions by default. Are there others?

pbatard, 2020-05-09:
> One solution is just to disable the virtual resolutions by default.

I would __strongly__ vote for that one, because this "non-native" being enabled by default has bothered me (and it seems quite a few others) for some time, resulting, in my case, in black bars and poor scaling on the monitor I perform most of my testing with, until I go to the menu to remove all the virtual res.

I understand the reason why you wanted to have them enabled, but it looks to me like we have been shifting the inconvenience of having to fiddle with the display options on what appears to be a much larger majority of users. Therefore, the law of inconveniencing the least amount of users dictates that we should disable the virtual resolutions by default and let the minority who need them re-enable them manually... even if you do happen to be in that minority. :wink:

Hopefully, when/if we split NVars from the firmware image, this will become less of an inconvenience as you should then be able to preserve virtual resolution settings when switching firmwares.

andreiw, 2020-05-09:
K, will send a DSC patch to disable, keeping only the native res by default. Tweaking this setting is definitely niche, and the current values get in the way of running any mainstream OS.

closed 2020-05-12 by andreiw