pftf/RPi4 #52: HTTP Boot cannot download 900MB ubuntu ISO

rgl, 2020-05-12:
At the server side:

```bash
wget http://cdimage.ubuntu.com/releases/20.04/release/ubuntu-20.04-live-server-arm64.iso
python3 -m http.server 8000
```

At the Pi side, with the v1.12 firmware, configuring HTTP Boot (e.g. http://192.168.1.69:8000/ubuntu-20.04-live-server-arm64.iso), and booting, it randomly fails at different downloading percentages, e.g.:

[image]

andreiw, 2020-05-12:
Happy to tell you this is fixed now. Wait for another release :)

andreiw, 2020-05-12:
https://github.com/tianocore/edk2-platforms/commit/3698b912301b51df5a6823669edbeba72ce05f69

pbatard, 2020-05-13:
I'm afraid I'm not seeing this fixed so I'm reopening it... 😢

samerhaj, 2020-05-13:
I verified v1.13 RELEASE build works with HTTP boot of local copies of;

ubuntu-20.04-live-server-arm64.iso (915 MB)
SLE-15-SP1-Installer-DVD-aarch64-GM-DVD1.iso (660 MB)

pbatard, 2020-05-13:
Yes, there appears to be an HTTP/Server component to that issue as well, as I don't get the HTTP download failure if the content is served from the default python 3.x HTTP server running on Windows, but I do get it from an Apache 2.0 server running on an Armbian RK3399 based machine...

I'll see if I can test a few more Apache setups to see if it's Apache itself or that specific Armbian platform. All the regular HTTP clients I tested seem to have no trouble downloading from that server though...

rgl, 2020-05-13:
Interesting, here it seems to be working fine!

pbatard, 2020-05-13:
There's definitely an Armbian/Apache2 package component to it, as using `python3 -m http.server 80` instead of Apache to serve the exact same file works fine, and similarly, serving the file from Apache2 running on a Debian 10.3 x86_64 platform works fine too.

So, though I suspect we might want to make edk2's HTTP GET support bit more resilient to discarded packets, since other clients (e.g. `wget` or regular browsers) can download the full 900MB image from that Apache2 server without trouble, the fact that this new issue is localized to a very specific platform/server combination would indicate that the core issue that was being tracked here had indeed been fixed.

I'll see if I can investigate whether regular HTTP clients do get discarded packets from the same content, and, if that's the case, how these clients handle them, so that we might be able to increase the robustness of our HTTP downloads.

closed 2020-05-13 by pbatard

pbatard, 2020-05-13:
For the sake of other folks possibly running into this, I'm just gonna add that the issue I observed is due to Armbian and possibly other RK399 compatible distros needing TCP offloading disabled for large transfers to work properly ([1](https://forum.armbian.com/topic/13544-nanopi-m4-and-other-rk3399-need-tcpudp-offloading-disabled/), [2](https://unix.stackexchange.com/questions/494290/nanopi-m4-rk3399-apache2-ssl-large-download-hangs/495378#495378)).

Basically, if ` tcpdump -i eth0 -vvv tcp | grep incorrect` produces loads of checksum errors on the platform you serve content from, you may want to look into disabling offloading with something like `ethtool -K eth0 rx off tx off`.

bzy-080408, 2020-05-15:
I tried to install ubunntu20.04 and it turned out i was successful.

bzy-080408, 2020-05-15:
I'm not using a network installation disc image

rgl, 2020-05-15:
@bzy-080408 How did you install Ubuntu?

bzy-080408, 2020-05-15:
Go to http://www.releases.ubuntu.com/20.04/download aarch64, burn to a USB stick, use uefi boot, and install (as with x86 installation)

bzy-080408, 2020-05-15:
I'll create a website to save the problems I've encountered and write the detailed installation method

bzy-080408, 2020-05-15:
https://github.com/bzy-080408/ubuntu-on-raspberrypi4B/