pftf/RPi4 #54: smbios: let the user configure the smbios asset tag from a uefi variable

rgl, 2020-05-18:
I want to be able to configure the SMBIOS TYPE 2 table Asset Tag from an uefi variable.

I have a WIP branch at https://github.com/rgl/edk2-platforms/tree/rgl-add-smbios-asset-tag-variable.

Can you please help getting it in good shape? :-)

I'm testing this with ipxe as described at https://github.com/rgl/raspberrypi-uefi-edk2-vagrant#ipxe.

But for some reason when iPXE boots with the following script, its `asset` variable is never `XXX`. Do you have any idea why?

```
echo Asset........ ${smbios/asset}
```

The complete ipxe script is at https://github.com/rgl/raspberrypi-uefi-edk2-vagrant/blob/master/rpi.ipxe#L11

rgl, 2020-05-18:
Oh I'm now reading the code at https://github.com/ipxe/ipxe/blob/e7f67d5a4c6e9f06aa7a9db1b4245f5e16f00bb2/src/interface/smbios/smbios_settings.c#L238-L247 and its actually using the smbios_enclosure_information table instead! will try that!

rgl, 2020-05-19:
I've now simplified to code to just use the `AssetTag` (name open for discussion) using the [EFI_GLOBAL_VARIABLE vendor guid](https://github.com/tianocore/edk2/blob/3806e1fd139775610d8f2e7541a916c3a91ad989/MdePkg/Include/Guid/GlobalVariable.h#L14).

But I'm now trying to set the variable value from the EFI shell, but it prevents me from doing that:

```
Shell> setvar AssetTag -bs -rt -nv =S"Hello World" =0x00
setvar: Unable to set - 8BE4DF61-93CA-11D2-AA0D-00E098032B8C - AssetTag
```

Do you known why?

Does the variable needs to be created first by the code?

samerhaj, 2020-05-19:
the EFI_GLOBAL_VARIABLE vendor guid is reserved for UEFI Spec defined variables. You cannot create variables using this GUID that are non-standard (not defined in the spec)

rgl, 2020-05-19:
@samerhaj that was it! thank you!

I can now set the asset tag with (the guid has to be uppercase or it does not actually get set for some reason):

```
setvar AssetTag -guid 7EEE1209-AE99-483C-9768-1224219FC0DA -bs -rt -nv =S"Hello World" =0x00
```

Can you please review https://github.com/rgl/edk2-platforms/tree/rgl-add-smbios-asset-tag-variable? :-)

samerhaj, 2020-05-20:
It may be better to do this in HII/UI setup menu. Those settings which are saved in UEFI variables, and Smbios code can extract it from there.

Look at SetupVariables in ConfigDxe for existing HII settings. These settings are saved as UEFI variables, which can be scripted in the Shell.

See for example: "RamLimitTo3GB".

The variable "RamLimitTo3GB" comes form the HII setting in ConfigDxeHii.vfr (the strings are defined in ConfigDxeHii.uni)

```
    efivarstore ADVANCED_RAM_MORE_THAN_3GB_VARSTORE_DATA,
      attribute = EFI_VARIABLE_BOOTSERVICE_ACCESS | EFI_VARIABLE_RUNTIME_ACCESS | EFI_VARIABLE_NON_VOLATILE,
      name  = RamMoreThan3GB,
      guid  = CONFIGDXE_FORM_SET_GUID;
```

During startup, PCDs are initialized from Variable in ConfigDxe `SetupVariable()`:

```
   PcdSet32 (PcdRamMoreThan3GB, 1);

    Size = sizeof (UINT32);
    Status = gRT->GetVariable (L"RamLimitTo3GB",
                               &gConfigDxeFormSetGuid,
                               NULL, &Size, &Var32);
    if (EFI_ERROR (Status)) {
      PcdSet32 (PcdRamLimitTo3GB, PcdGet32 (PcdRamLimitTo3GB));
    }
  } else {
    PcdSet32 (PcdRamMoreThan3GB, 0);
    PcdSet32 (PcdRamLimitTo3GB, 0);
  }
```
Then the PCD is used in actual code that consumes it, such as `MemArrMapInfoUpdateSmbiosType19()` when building SMBIOS

rgl, 2020-05-20:
The initial rough cut is at https://github.com/rgl/edk2-platforms/tree/rgl-add-smbios-asset-tag.

I'm surely doing something wrong because after I set the asset tag to `hello` and I try to save setting with `F10` it fails on me:

```
/------------------------------------------------------------------------------\
|                           Advanced Configuration                             |
\------------------------------------------------------------------------------/

   Smbios Enclosure Asset Tag hello
   Limit RAM to 3 GB          <Enabled>
   System Table Selection     <ACPI>
  /--------------------------------------------------------------------------\
  |                                                                          |
  |               Submit Fail For Form: Advanced Configuration.              |
  |Press D(d) to discard changes for this form, Press G(g) to go to this form|
  |                                                                          |
  \--------------------------------------------------------------------------/

/------------------------------------------------------------------------------\
|                         F9=Reset to Defaults      F10=Save                   |
| ^v=Move Highlight       <Enter>=Select Entry      Esc=Exit                   |
\------------------------------------------------------------------------------/
                                                    Configuration changed
```

Do you have idea why?

I'm not really sure how to declare the PCD in Platform/RaspberryPi/RaspberryPi.dec, maybe thats what is fubar.

samerhaj, 2020-05-27:
I pushed a working change to: https://github.com/pftf/edk2-platforms/tree/assettag_dev1

Default value is shown in the UI, NV Variable, and SMBIOS:

[image]

```
Shell> dmpstore AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5
Variable NV+RT+BS 'CD7CC258-31DB-22E6-9F22-63B0B8EED6B5:AssetTag' DataSize = 0x2
2
  00000000: 20 00 20 00 20 00 20 00-20 00 20 00 20 00 20 00  * . . . . . . . .*
  00000010: 20 00 20 00 20 00 20 00-20 00 20 00 20 00 20 00  * . . . . . . . .*
  00000020: 00 00                                            *..*

Shell> smbiosview -t 2
...
AssetTag:
...

Shell> smbiosview -t 3
...
AssetTag:
...
```

Changing the value is the UI is also reflected in the UEFI NV variable and SMBIOS:

[image]

```

Shell> dmpstore AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5
Variable NV+RT+BS 'CD7CC258-31DB-22E6-9F22-63B0B8EED6B5:AssetTag' DataSize = 0x2
2
  00000000: 30 00 31 00 32 00 33 00-34 00 35 00 36 00 37 00  *0.1.2.3.4.5.6.7.*
  00000010: 38 00 39 00 61 00 62 00-63 00 64 00 65 00 66 00  *8.9.a.b.c.d.e.f.*
  00000020: 00 00                                            *..*
Shell> smbiosview -t 2
...
AssetTag: 0123456789abcdef
...

Shell> smbiosview -t 3
...
AssetTag: 0123456789abcdef
...

samerhaj, 2020-05-27:
Pushed change without using the PCD. This looks better. Default value is an empty string. Max size is 32 characters.
https://github.com/pftf/edk2-platforms/commit/127957c59709d92baacede6fdd37ad7b095da80a

Default:

[image]

```
Shell> dmpstore AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5

Variable NV+RT+BS 'CD7CC258-31DB-22E6-9F22-63B0B8EED6B5:AssetTag' DataSize = 0x42
  00000000: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000010: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000020: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000030: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000040: 00 00                                            *..*

Shell> smbiosview -t 2
...
AssetTag:
...

Shell> smbiosview -t 3
...
AssetTag:
```

Setting to max string size:

[image]

```
Shell> dmpstore AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5
Variable NV+RT+BS 'CD7CC258-31DB-22E6-9F22-63B0B8EED6B5:AssetTag' DataSize = 0x42
  00000000: 30 00 31 00 32 00 33 00-34 00 35 00 36 00 37 00  *0.1.2.3.4.5.6.7.*
  00000010: 38 00 39 00 61 00 62 00-63 00 64 00 65 00 66 00  *8.9.a.b.c.d.e.f.*
  00000020: 30 00 31 00 32 00 33 00-34 00 35 00 36 00 37 00  *0.1.2.3.4.5.6.7.*
  00000030: 38 00 39 00 61 00 62 00-63 00 64 00 65 00 66 00  *8.9.a.b.c.d.e.f.*
  00000040: 00 00                                            *..*

Shell> smbiosview -t 2
...
AssetTag: 0123456789abcdef0123456789abcdef
...

Shell> smbiosview -t 3
...
AssetTag: 0123456789abcdef0123456789abcdef
...

```

rgl, 2020-05-27:
Why are some variables `CHAR16     AssetTagVar[128] = L"";` using 128? Can't they use `ASSET_TAG_STR_STORAGE_SIZE`?

rgl, 2020-05-27:
I confirm that its working fine here too! :-)

For reference, I've tested setting from command line with a maximum length string with:

```
setvar AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5 -bs -rt -nv =L"CD7CC25831DB22E69F2263B0B8EED6B5" =0x0000
```

And successfully read it back:

```
dmpstore AssetTag -guid CD7CC258-31DB-22E6-9F22-63B0B8EED6B5
```

```
Variable NV+RT+BS 'CD7CC258-31DB-22E6-9F22-63B0B8EED6B5:AssetTag' DataSize = 0x42
  00000000: 43 00 44 00 37 00 43 00-43 00 32 00 35 00 38 00  *C.D.7.C.C.2.5.8.*
  00000010: 33 00 31 00 44 00 42 00-32 00 32 00 45 00 36 00  *3.1.D.B.2.2.E.6.*
  00000020: 39 00 46 00 32 00 32 00-36 00 33 00 42 00 30 00  *9.F.2.2.6.3.B.0.*
  00000030: 42 00 38 00 45 00 45 00-44 00 36 00 42 00 35 00  *B.8.E.E.D.6.B.5.*
  00000040: 00 00                                            *..*
```

With EDK Setup too:

```
/------------------------------------------------------------------------------\
|                           Advanced Configuration                             |
\------------------------------------------------------------------------------/

                                                         Set the system Asset
   Limit RAM to 3 GB          <Enabled>                  Tag
   System Table Selection     <ACPI>
   Asset Tag                  CD7CC25831DB22E69F2263B0B8
                              EED6B5

```

And with iPXE too:

```
iPXE 1.20.1 (g8f151) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP HTTPS iSCSI SRP EFI Menu
Platform..... efi
Architecture. arm64
Product...... Raspberry Pi 4 Model B
Manufacturer. Sony UK
Asset........ CD7CC25831DB22E69F2263B0B8EED6B5

```

samerhaj, 2020-06-12:
Fix is pushed upstream

closed 2020-06-12 by samerhaj