pftf/RPi4 #55: SecureBoot enablement

samerhaj, 2020-05-27:
We should enable SecureBoot on the RPi4.

This is important as an example / experiments / testing environment, even if the variable store is not secure (as it is currently in RPI FD file on the SD card), and even after the variables move to the SPI storage (per https://github.com/pftf/RPi4/issues/6)

jlinton, 2020-09-29:
I haven't tried secure booting the rpi, but secure boot is enabled in the public firmware builds (-DSECURE_BOOT_ENABLE=TRUE), although I don't think (or didn't see when I looked at it a couple weeks ago) that any keys are enrolled.

samerhaj, 2020-09-29:
There are no keys enrolled by default in EDK2 platforms. One reference example for doing this is OVMF, which includes an application to enroll default UEFI keys:(https://github.com/tianocore/edk2/tree/master/OvmfPkg/EnrollDefaultKeys). for now, the keys/certs are hard-coded in https://github.com/tianocore/edk2/blob/master/OvmfPkg/EnrollDefaultKeys/AuthData.c and include MS KEK CA, MS Windows Production PCA, and the MS UEFI CA, as well as a blank DBX.

All of these are generic enough and could be moved from OVMF to a more common location in edk2 for use by any platform. The code could be improved to move the keys/certs from being hard-coded in AuthData to auto-generated at build time (python tool to convert certs into autogen AuthData C arrays)

jlinton, 2020-09-29:
The EnrollDefaultKeys.efi utility can be built for armvirt as well, then it just needs to be run against the firmware. I've had various sets of patches to do that for fedora/edk2 for a couple years now (https://src.fedoraproject.org/fork/jlinton/rpms/edk2/c/b0c7672e5a5753a7b135237ee165f1b2e98264a2) but there is ongoing push back about the security of actually doing it.

samerhaj, 2020-11-16:
Adding a TODO from https://github.com/pftf/RPi4/issues/11#issuecomment-700799247
"we need to assure that the JTAG pins are "disabled" if secure boot is enabled"

samerhaj, 2021-04-16:
There is an RFC sent to EDK2 to add "default SecureBoot keys" support. We could do this on the RPi4 as well:

https://edk2.groups.io/g/rfc/topic/edk2_devel_rfc_secure/82139806