pftf/RPi4 #58: gpu_mem=16 doesn't work (missing start4cd.elf and fixup4cd.dat)

andreiw, 2020-05-30:
When gpu_mem=16 is set in config.txt, the "cut down" version of VPU fw is loaded (e.g. without codecs and other bloat) - start4cd (start_cd for non-Pi4) and fixup4cd (fixup_cd for non-Pi 4).

I suggest distributing the full thes of *.elf and *.dat files with our ZIP-packages releases.

See https://github.com/raspberrypi/firmware/issues/959

pbatard, 2020-05-30:
I don't think we want to do that.

The firmware archive is meant to contain only the files it needs for the `config.txt` provided (the only exception being the miniuart overlay for the Pi3, but that's because we explicitly document how to switch UARTs in the EDK2 docs) and that's it.

You do understand that if we start going that route then logically, we have to provide __all__ of the `.dtbo` overlays from https://github.com/raspberrypi/firmware/tree/master/boot/overlays, because is we do start to cater to people using whatever possible option they can have in their `config.txt` then we have no other logical choice than see it all the way through.

In other words, we can't just cherry pick files that are relevant to someone's wish to use a specific `config.txt` option and not automatically include all of the other ones.

Hence the reasoning of __not__ including any of the firmware files that are not relevant to the default `config.txt` that comes with the archive. And since `gpu_mem=16` is not being used, I see no reason to add `start4cd`. Users who do want to use `gpu_mem=16` should provide that file themselves.

I also don't expect many users to have a requirement for this, which is another reason not to include it. If it was likely that 1% of the people downloading the zip were going to use `gpu_mem=16` eventually, then it might make sense to add it. But one can be pretty confident that this is not going to be the case.

andreiw, 2020-05-30:
That’s fair. How about we mandate gpu_mem=16?

pbatard, 2020-05-30:
I'm not sure I see the point of doing so. If there's a majority for it, sure, but I'd rather stick with the usual "element of least surprise" and keep `config.txt` in line with what one would expect for a regular non UEFI boot.

What exactly do we think we have to gain from using `gpu_mem=16`, that's not going to incur us having to explain _"Why are you guys using `gpu_mem=16` and `start4cd.elf`?"_ on semi regular basis?

In other words, as far as I'm concerned, the less original we try to be with `config.txt` and the firmware binaries, the better. But that's just my vote.

andreiw, 2020-06-01:
Got it. Fair enough.

closed 2020-06-01 by andreiw

kpfleming, 2021-06-17:
I can understand the reasoning here, but I'm thankful that @andreiw opened this and I looked through the closed issues :-) Maybe it would be useful to have an FAQ to provide guidance for not-common but also not-rare configurations like this (and also things like `disable-bt` and `disable-wifi` overlays)?

Nebucatnetzer, 2024-01-30:
I just discovered this issue and disable Bluetooth and Wi-Fi on all my Pis.
Can I just put the files to the right place and it works?