pftf/RPi4 #60: Pi 4G memory recognition error

bzy-080408, 2020-06-03:
My is the magnesium light 4G memory version, when I use the official 32-bit system is to show memory has 3902MB, and when the 3072MB memory limit is turned off the display memory is 4096MB when I use a third-party 64-bit system also shows 3902MB. Hope to fix it as soon as possible.

pbatard, 2020-06-03:
Can you please be clearer about what you want?

You do realize that even when 4GB of RAM are available, the GPU needs to reserve some of that RAM for Video, so it becomes unavailable to the system.

This is the same as for a PC with an integrated GPU: You may have 8 GB of RAM available, but if you reserve 512 MB for the video, Windows or Linux will only have 7.5 GB available.

Therefore, I don't believe that there is anything for us to "fix" here.

If you want to try to reclaim memory, please look into the [`gpu_mem` `config.txt` option](https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md), knowing that you will also have to provide a different version of `start4.elf` and possibly some other files.

As I believe I have answered your question, and I don't see anything actionable for the UEFI firmware, I will close this issue.

closed 2020-06-03 by pbatard

bzy-080408, 2020-06-03:
Do you mean that those memory is consumed by the graphics card?

pbatard, 2020-06-03:
Yes.

bzy-080408, 2020-06-03:
And can you tell me where you're from? If we use the same language, I no longer need to translate.

pbatard, 2020-06-03:
Other people may be interested in this conversation, so I will kindly request that you use English if you have support requests.

bzy-080408, 2020-06-03:
OK