pftf/RPi4 #63: Hibernate/wakeup support

jlinton, 2020-06-06:
I don't see a reason we can't support S4 hibernate in a rough format (wakeup via power on). Fedora seems to be mostly there, but resume isn't working correctly at the moment. Frankly, i'm not sure this isn't at least partially a fedora bug, or if nothing else PEBKAK. 

Of course being able to select a wakup timeout will require a rtc that can wake the machine, and I don't yet see how to make that work either (even with a rtc hat) short of "simulating" it via going into atf/uefi and sleeping.

Lets keep this as a TODO, and if I get a chance I will take a look at it, sufficiently that something like `systemctl hibernate` and a power cycle work.

cleverca22, 2020-06-08:
ive also got a theory on how to implement suspend to ram with the open firmware

in theory, you could just reverse all of the "enable arm cores" steps, to go back to a lower power state, but leave the dram on, and then at some later time, turn them back on

this also leads to an idea about how you might do hibernate...
any time you tell an rpi to "shutdown", it actually just reboots
for an rpi4, the bootcode.bin in the eeprom detects a magic flag, and will then halt (either by just running a sleep opcode, or by cutting power)

in theory, you could shove a custom `recovery.bin` onto the fat32 partition, and when you reboot, that runs (in a low-power state)
it can then use the VPU timers to count to X, then delete itself and reboot again, causing the official firmware to boot uefi again, and resume the OS

jlinton, 2021-12-09:
Well this mostly works in F35 (by mostly I mean it doesn't). It does, however, write everything to the swap partition, and run the power off sequence. Then after manually power cycling the pi, it boots, detects the hibernation,  and reads the image into memory. Then 5.16 crashes while copying pages around. So, I tried duplicating this on a couple of other arm platforms and haven't even found one that gets as far as the pi, the failure though looks like a generic arm64 failure rather than anything pi specific (even qemu/kvm isn't working at the moment).