pftf/RPi4 #64: Debian installer uses serial console by default

pbatard, 2020-06-08:
It looks like, currently, the console from an uninitialized firmware is neither completely serial nor graphical, as, if you try to install Debian 11, and if you don't go to the console settings, it will default to using serial console instead of graphical.

Most likely, other distros' installeres also default to using serial.

This is of course very problematic for users as, if they don't explicitly go to the setup/console preference selection, they are only going to be met with a black screen.

It would appear that the console settings we apply when we detect that no console has been set (which result in boot message `ConsolePrefDxeEntryPoint: no console preference found, defaulting to graphical`) and the ones we apply when graphical has been explicitly do not result in the same outcome...

pbatard, 2020-06-13:
Further investigation shows that, until you explicitly go into the menu to set the variable, the [`OnReadyToBoot()` event from `ConsolePrefDxe.c`](https://github.com/tianocore/edk2/blob/9af1064995d479df92e399a682ba7e4b2fc78415/EmbeddedPkg/Drivers/ConsolePrefDxe/ConsolePrefDxe.c#L282-L285) never happens, even as the `gBS->CreateEventEx` call reports success...

And it's not a matter of the `OnReadyToBoot()` being called and bailing out early.
`OnReadyToBoot()` is just never called, even as its associated event was successfully queued up.

pbatard, 2020-06-15:
The best explanation I have for this behaviour is that, until we save the variables, we default to booting `PlatformRecovery`, and therefore the `EFI_EVENT_GROUP_READY_TO_BOOT` event group is not being processed as, per specs, this is tied to Boot Manager being invoked:

> `EFI_EVENT_GROUP_READY_TO_BOOT`
> This event group is notified by the system when the Boot Manager is about to load and execute a
boot option.

For reference this is the debug output from a firmware where the variables have not been saved:

```
[Bds]=============Begin Load Options Dumping ...=============
  Driver Options:
  SysPrep Options:
  Boot Options:
    Boot0000: UiApp              0x0109
    Boot0001: UEFI Shell                 0x0000
  PlatformRecovery Options:
    PlatformRecovery0000: Default PlatformRecovery               0x0001
[Bds]=============End Load Options Dumping=============
[Bds]BdsWait ...Zzzzzzzzzzzz...
[Bds]BdsWait(5)..Zzzz...
.[Bds]BdsWait(4)..Zzzz...
.[Bds]BdsWait(3)..Zzzz...
.[Bds]BdsWait(2)..Zzzz...
.[Bds]BdsWait(1)..Zzzz...
..[Bds]Exit the waiting!
Process PlatformRecovery0000 (Default PlatformRecovery) ...
```

And the same output after they have:
```
[Bds]=============Begin Load Options Dumping ...=============
  Driver Options:
  SysPrep Options:
  Boot Options:
    Boot0000: UiApp              0x0109
    Boot0001: UEFI Shell                 0x0000
    Boot0002: SD/MMC on Arasan SDHCI             0x0001
    Boot0003: UEFI SanDisk USB Extreme Pro 00000000000000000000                  0x0001
    Boot0004: UEFI PXEv4 (MAC:DCA6320C92F5)              0x0001
    Boot0005: UEFI PXEv6 (MAC:DCA6320C92F5)              0x0001
    Boot0006: UEFI HTTPv4 (MAC:DCA6320C92F5)             0x0001
    Boot0007: UEFI HTTPv6 (MAC:DCA6320C92F5)             0x0001
  PlatformRecovery Options:
    PlatformRecovery0000: Default PlatformRecovery               0x0001
[Bds]=============End Load Options Dumping=============
[Bds]BdsWait ...Zzzzzzzzzzzz...
[Bds]BdsWait(5)..Zzzz...
.[Bds]BdsWait(4)..Zzzz...
.[Bds]BdsWait(3)..Zzzz...
.[Bds]BdsWait(2)..Zzzz...
.[Bds]BdsWait(1)..Zzzz...
..[Bds]Exit the waiting!
[Bds]Stop Hotkey Service!
[Bds]UnregisterKeyNotify: 000B/0000 Success
[Bds]UnregisterKeyNotify: 0017/0000 Success
[Bds]UnregisterKeyNotify: 0000/000D Success
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>><<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
>>>>>>>>>>>>>>> OnReadyToBoot: OnReadyToBoot <<<<<<<<<<<<<<<
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>><<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
```

pbatard, 2020-06-16:
EDK2 patch proposal in https://edk2.groups.io/g/devel/message/61327 + https://edk2.groups.io/g/devel/message/61328.

samerhaj, 2020-06-19:
Seems like an EDK2 bug : https://edk2.groups.io/g/devel/message/61433

samerhaj, 2020-08-26:
discussion in BZ issue https://bugzilla.tianocore.org/show_bug.cgi?id=2831

baryluk, 2021-02-05:
FYI. I tried yesterday Debian installer for bullseye alpha3, https://cdimage.debian.org/cdimage/bullseye_di_alpha3/arm64/iso-dvd/debian-bullseye-DI-alpha3-arm64-DVD-1.iso specifically, with RPi4 UEFI 1.22 on RPi 4 8GB, and the installation was by default graphical (I mean "not over serial"). I didn't even need to adjust kernel parameters or anything, it was this way by default.