pftf/RPi4 #65: ipxe network broken in v1.15?

rgl, 2020-06-13:
[v1.15](https://github.com/pftf/RPi4/releases/tag/v1.15) somehow broke the ipxe network initialization in my RPi4 8GB.

In v1.14 (in a RPi4 **4**GB) the ipxe `net0` device initialized fine, but with v1.15 the initialization process does not seem to happen has before, and ipxe fails to find/initialize/use the network.

To workaround this, it seems that using the ipxe `ifstat` command triggers a crash, but something (uefi?) is immediately restarting ipxe, and somehow it works the second time. Here's the output:

```plain
ESC (setup), F1 (shell), ENTER (boot).iPXE initialising devices...ok

iPXE 1.20.1 (g8f151) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP SRP EFI Menu
Platform..... efi
Architecture. arm64
Product...... Raspberry Pi 4 Model B
Manufacturer. Sony UK
Asset........ PI00000001
UUID......... 00d03114-0000-0000-0000-dca632b0ba1d
Serial....... 0000DCA632B0BA1D
Network interfaces:

  Error: Could not detect network connection.

  Error: Could not detect network connection.
iPXE initialising devices...ok

iPXE 1.20.1 (g8f151) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP SRP EFI Menu
Platform..... efi
Architecture. arm64
Product...... Raspberry Pi 4 Model B
Manufacturer. Sony UK
Asset........ PI00000001
UUID......... 00d03114-0000-0000-0000-dca632b0ba1d
Serial....... 0000DCA632B0BA1D
Network interfaces:
net0: dc:a6:32:b0:ba:1d using SNP on SNP-0x35c0c598 (closed)
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (http://ipxe.org/1a086194)]
Configuring net0 dc:a6:32:b0:ba:1d (SNP) from dhcp...
Waiting for link-up on net0...... ok
Configuring [dhcp] (net0 dc:a6:32:b0:ba:1d)...... ok
```

rgl, 2020-06-13:
Hummm everything works as expected when the memory is limited to 3GB.

samerhaj, 2020-06-17:
I see link status/detection errors. 

We have some fixes for the GENET driver media detection code, that may be related.

samerhaj, 2020-06-17:
Can you please pull this patch and see whether it resolves this issue or not?

https://edk2.groups.io/g/devel/message/61435

samerhaj, 2020-06-17:
Fix committed to tianocore/edk2@2d233af64b8f73d1b1e138b302e6344f7c2e0f4e

Will port to RPi PlatformBootManagerLib and send a patch

closed 2020-06-18 by pbatard

rgl, 2020-06-19:
I've tried v1.16 and the iPXE network initialization now works at the first. Thanks for looking into this! And, sorry for not being able to test this sooner.

commit 747d06f6 (2020-06-18): Update EDK2 to latest

* Closes #65 through tianocore/edk2-platforms@8dd78ea11a38d2d7e8031c4783e9f2ca5569956b
* Addresses part of #75 through tianocore/edk2-platforms@d2c797e22184028cf3a8f3e75737c75ef917b422