pftf/RPi4 #66: keyboard and mouse not working

Jeshzz, 2020-06-16:
keyboard and mouse not working while booting into windows  with firmware 1.15 and windows 10 18362.267 from worproject.ml. 
keyboard and mouse work fine at setup can able go to device manager, boot manager etc. however while windows loading/loaded then keyboard and mouse not working.

Using a type c hub with external power.

pbatard, 2020-06-16:
From the readme:

> You will NOT be able to use any of the USB-A ports if you try to run Windows

Are you using the USB-A ports from the Pi 4? If so that will not work because Microsoft xHCI driver has an issue with the xHCI controller that the Pi 4 uses (and this is an issue that cannot be fixed by the UEFI firmware).

Jeshzz, 2020-06-16:
No I am using type c hub/ adapter.

pbatard, 2020-06-16:
So is it a regression? I.e. did it work prior to version 1.15?

Because if not, then this seems like a Windows support issue with your specific hardware. As opposed to what you may be led to think, once UEFI exits and Windows has booted, it's the Windows native Microsoft drivers that control __ALL__ USB access, and UEFI is not used at all.

So there again, if the Windows drivers have a limitation, there's not much we can do about that.

Please bear in mind that, until Microsoft updates Windows to properly support the Pi 4 hardware, or someone writes their own Windows driver to compensate for the current Windows limitations, you shouldn't really expect Windows to work as well as if you were using a PC.

Without Microsoft's direct involvement to improve hardware support, you are very unlikely to find that Windows 10 is as usable as you wish on the Pi 4, if usable at all...

Biswa96, 2020-06-16:
Off-topic. @Jeshzz May you provide any example of that hardware? I can't find any or maybe don't know the appropriate name.

andreiw, 2020-07-11:
There have been no DWC2 driver changes in UEFI since forever, and your devices clearly work in UEFI... so... what's the bug?

You need MCCI drivers for the Type C. It will not work without MCCI drivers for WIndows.

Closing this as it is not apparently UEFI has anything to do with it.

closed 2020-07-11 by andreiw