pftf/RPi4 #67: ACS FWTS ACPI DSDT failures

samerhaj, 2020-06-17:
From the ACS tests on Rel 1.9

https://github.com/pftf/acs-reports/blob/2ec70f92201f5f4ab36006ba36c500f3206b1958/v1.9/acs-enterprise-v1.6/luv-results-2020-04-21--14-08-49/parsed/fwts#L128

```
1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.USB0._CID returned a
1.1  fwts string INFO 'DWC_OTG' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.USB0._CID returned a
1.1  fwts string INFO 'DWC2_OTG' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.GPU0._CID returned a
1.1  fwts string INFO 'VC4' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.RPIQ._CID returned a
1.1  fwts string INFO 'RPIQ' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.VCIQ._CID returned a
1.1  fwts string INFO 'VCIQ' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.VCSM._CID returned a
1.1  fwts string INFO 'VCSM' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.GPI0._CID returned a
1.1  fwts string INFO 'BCMGPIO' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.I2C1._CID returned a
1.1  fwts string INFO 'BCMI2C' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.I2C2._CID returned a
1.1  fwts string INFO 'BCMI2C' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.SPI0._CID returned a
1.1  fwts string INFO 'BCMSPI0' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.SPI1._CID returned a
1.1  fwts string INFO 'BCMAUXSPI' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.URTM._CID returned a
1.1  fwts string INFO 'MINIUART' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.SDC1._CID returned a
1.1  fwts string INFO 'ARASAN' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts FAILED INFO [MEDIUM] MethodCIDInvalidString: Test 6, \_SB_.GDV0.SDC2._CID returned a
1.1  fwts string INFO 'SDHST' but it was not a valid PNP ID or a valid ACPI ID.

1.1  fwts 134 INFO passed, 14 failed, 6 warnings, 0 aborted, 184 skipped, 0 info only.
```

samerhaj, 2020-08-13:
the invalid _CID issue due to a check in FWTS  (https://git.launchpad.net/fwts/tree/src/lib/src/fwts_acpi_object_eval.c#n1924 ) that validates the _CIDs and _HIDs to have valid format per ACPI spec. The test simply checks for either a PNP ID (3 capitals followed by 4 hex) or ACPI ID (4 capitals or digits followed by 4 hex), as defined in the ACPI spec (*). Removing these _CIDs or changing them to valid strings resolves the issue. However, these Ids are used in the windows IoT BSP drivers https://github.com/ms-iot/rpi-iotcore and removing them will break those drivers to some users:

BCMGPIO
BCMI2C 
BCMSPI0 
BCMAUXSPI 
RPIQ
DWC_OTG
DWC2_OTG
VC4
VCSM
VCIQ
MINIUART
ARASAN
SDHST

driver1998, 2020-08-19:
Microsoft already defined standard-looking IDs for _HID, so we can just use those for _CID as well and deprecate the current non-standard ones. (Or just remove these _CIDs)

Changes for open source drivers are easy (actually most drivers match using _HID only or as an option, except Arasan/SDHost). But the proprietary MCCI DWCHSOTG driver matches with DWC_OTG only, so that would be tricky (either convince MCCI to release a v2, or hack the INF files).

Side note: For the Arasan controller, I need to somehow separate Pi3 and Pi4 in ACPI (the ACT LED is wired differently), if _HID is a good way to do this, I propose the following IDs:

Pi3
_HID RPI32847
_CID BCM2847

Pi4
_HID RPI42847
_CID BCM2847

samerhaj, 2020-08-26:
Moved the _AEI failure to its own issue

samerhaj, 2021-08-05:
Fixed released in https://github.com/pftf/RPi4/releases/tag/v1.29

closed 2021-08-05 by samerhaj