pftf/RPi4 #72: CS SCT Memory AllocatePages failures

samerhaj, 2020-06-18:
https://raw.githubusercontent.com/pftf/acs-reports/master/v1.15/acs-enterprise-v2.4/SCT_RUN/Overall/Summary.log

```
S.AllocatePages - back range -- FAILURE
3E0A81A9-3670-4239-8C91-5D99613D9644
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1115:Status - Not Found, TPL - 4, MemType - 5
```

```
BS.AllocatePages - back range return Address -- FAILURE
0016743C-47D3-46EF-AAC6-3B53872703B1
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1156:TPL - 4, MemType - 5
```

```
BS.AllocatePages - back range -- FAILURE
3E0A81A9-3670-4239-8C91-5D99613D9644
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1115:Status - Not Found, TPL - 4, MemType - 6
```

```
BS.AllocatePages - back range return Address -- FAILURE
0016743C-47D3-46EF-AAC6-3B53872703B1
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1156:TPL - 4, MemType - 6
```

```
BS.AllocatePages - back range -- FAILURE
3E0A81A9-3670-4239-8C91-5D99613D9644
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1115:Status - Not Found, TPL - 4, MemType - 9
```

```
BS.AllocatePages - back range return Address -- FAILURE
0016743C-47D3-46EF-AAC6-3B53872703B1
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1156:TPL - 4, MemType - 9
```
```
BS.AllocatePages - back range -- FAILURE
3E0A81A9-3670-4239-8C91-5D99613D9644
/home/edhcha01/RELEASE_V2.4/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.1+gitAUTOINC+b558bad254-r0/edk2/SctPkg/TestCase/UEFI/EFI/BootServices/MemoryAllocationServices/BlackBoxTest/MemoryAllocationServicesBBTestFunction.c:1115:Status - Not Found, TPL - 4, MemType - 10
```
 
etc..
```
AllocatePages_Func: [FAILED]
  Passes........... 1368
  Warnings......... 0
  Errors........... 24
```

paulyang0125, 2020-07-16:
can try SCT by this ACS pre-built image that includes most up-to-date fixes for sct and sbsa so far 

https://drive.google.com/drive/folders/1lEf78-doK64UgBLCw7bx1ODoyUe3B2Dp?usp=sharing

samerhaj, 2020-07-16:
Repeated the test using v1.17 FW and the ACS 2.4_plus latest build, and these tests now pass:

**Overall logs:**
https://raw.githubusercontent.com/pftf/acs-reports/master/v1.17/acs-enterprise-v2.4%2B/SCT_RUN/Overall/Summary.log

**Specific logs:** 
https://github.com/pftf/acs-reports/blob/master/v1.17/acs-enterprise-v2.4%2B/SCT_RUN/Log/BootServicesTest/MemoryAllocationServicesTest0/AllocatePages_Func_0_0_21338335-67BA-4392-8D5E-7184C6D601A6.log

**Results**
```
AllocatePages_Func: [PASSED]
  Passes........... 1392
  Warnings......... 0
  Errors........... 0
```

closed 2020-07-16 by samerhaj

samerhaj, 2020-08-26:
Confirmed fixed in v1.19 FW and ACS 2.5 beta:

https://github.com/pftf/acs-reports/blob/master/v1.19/acs-enterprise-v2.5-beta/SCT_RUN/Overall/Summary.log

```
AllocatePages_Func: [PASSED]
  Passes........... 1392
  Warnings......... 0
  Errors........... 0
```