pftf/RPi4 #76: Items required solving for Debian support

pbatard, 2020-06-19:
Just going to create an issue to gather all the items that require solving before we can propose Debian installation as a viable option:

### Major showstoppers

- [x] Linux kernel Genet ACPI bindings &rarr; Added to kernel 5.x and backported to the current Debian 11.0 kernel.
- [ ] UEFI boot should not freeze when not SD card is present &rarr; Needed since, with the latest EEPROM, the Pi 4 can boot directly to a GPT/ESP on a USB drive, which would allow copying the Debian 11 netinst ISO to an ESP on the target installation drive, along with the firmware files, and get through the whole installation process using one drive only. Tracked in #37.

### Partial showstoppers

These can be worked around, but asking users to do so is less than ideal, as it will greatly degrade their experience during installation. Therefore classified as showstoppers.

- [ ] Console should properly default to graphical &rarr; Tracked in #64
- [ ] Debian needs to remount `/cdrom` `rw` when it maps to an ESP. &rarr; Needed for the same reason as the above, as this is another hurdle for easy install through a single USB. Submitted as a Debian bug in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=963156

### Would be nice to have

These are not considered showstoppers as, once a user manages to complete a Debian netinst installation, they will get kernel updates and therefore will naturally see these fixes integrated once they have been upstreamed and backported.

- [ ] Have the kernel support RAM above 3 GB &rarr; Tracked in #20 
- [ ] Have the kernel support the SD controller &rarr; Tracked in #26

ehem, 2020-07-13:
> Added to kernel 5.x and backported to the current Debian 11.0 kernel.

Possible to get more specificity here?  I _suspect_ you're referring to Debian's linux-source-5.5 package (and presumably later versions)?  Perhaps linux-source-5.6 (and later)?

I would place memory above 3GB as rather more than merely "nice to have".  On a 4GB RP4, losing 25% of memory is a big hit, on a 8GB RP4 you're losing more than half of system memory.  That is darn near a major show-stopper.

pbatard, 2020-07-13:
> Possible to get more specificity here?

The Genet driver (or more exactly the ACPI binding patches for Genet), which is included in recent 5.x mainline kernels was backported into the semi-custom 5.y kernel that Debian 11 uses. If you pick the latest Debian 11 pre-release ISO, you will find that it has Genet support.

> I would place memory above 3GB as rather more than merely "nice to have". 

Sorry but this is not a show stopper.

It does not prevent you from running or installing Linux, and, if you have Debian installed with networking, you will get this fixed __automatically__ through system update. Therefore, not an actually problematic item.

dony71, 2020-07-17:
does analog audio work?
i have add dtparam=audio=on in config.txt

jlinton, 2020-11-16:
As of 5.10, the linux mcfg message has been reduced to pr_debug() which should make it less noticeable.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v5.10-rc4&id=d24e124577cdb47824c6665af16250807c2daa89

pbatard, 2020-11-16:
Great. Thanks for looking after this!

jlinton, 2021-02-22:
The edk2+linux-firmware pieces have landed, and the the ACPI enablement patch for linux is in -next. I would expect that will merge to mainline RSN, so its should now be a debian issue until 4.12 is picked up by debian.

https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/mmc/host/sdhci-iproc.c?h=next-20210222&id=4f9833d3ec8da34861cd0680b00c73e653877eb9

meskilla, 2021-03-28:
> does analog audio work?
> i have add dtparam=audio=on in config.txt

snd-bcm2835 does not add the alsa compat sink necessary to use the headphone jack. This problem goes away with newer *4*elf & *4*dat broadcom blobs or with a newer bcm2711-rpi-4-b.dtb than what atm supplied by the rpi4-uefi firmware. So the only way to have sound is get rid of the nice grub enviroment after installation and use the blobs directly :-( . Would be nice to see a fix of this issue: allow people to replace bcm2711-rpi-4-b.dtb and add start4x.dat/start4x.elf in order to enable vc4-fkms-v3d.dtbo

paulwratt, 2022-02-17:
"Console should properly default to graphical" has been resolved according to the [EDK2 bug 2831 conclusion](https://bugzilla.tianocore.org/show_bug.cgi?id=2831#22), and the resulting spec update that provides new variables.

> This means we can now update the TianoCore EDK2 / EDK2-Platforms code to signal EFI_EVENT_GROUP_READY_TO_BOOT (and EFI_EVENT_GROUP_AFTER_READY_TO_BOOT when it gets implemented) on platform recovery path, similar to what is done on normal boot path.

#64 should be able to move forward now.

paulwratt, 2022-02-17:
It appears the only items left of this "Debian FIXME" list are the WiFi Blob issues, and the "Nice to have" vc4/5 option.

Is it possible that the WiFi blob need DeviceTree to function?

Is vc4/5 a practical reality any time soon? (I doubt that, but I have also not researched it either)

hellosway, 2022-03-09:
Doesnt just need device tree, and there is a setting to enable devicetree. After that shouldnt it be just passing the .dtb file to the kernel.

jlinton, 2022-03-10:
@hellosway The device  tree support for the RPi4 wasn't the main focus area for this project. The idea was to create a single UEFI+ACPI firmware image that can boot a wide range of OSs rather than having ACPI for some of them and DT for others. Mostly because DT is not standard, and its not even well specified as the linux devs are constantly tweaking it, which is why there are a few dozen differing DT's for the base bcm2711 SoC floating around, with varying levels of support depending on the rpi foundations kernel, mainline, etc.

hellosway, 2022-03-11:
> @hellosway The device tree support for the RPi4 wasn't the main focus area for this project. The idea was to create a single UEFI+ACPI firmware image that can boot a wide range of OSs rather than [...]

I understand that, but adding a dtoverlay to config.txt with vc4 doesnt pass it to linux when using this firmware. When using uboot, the devicetree is passed to the kernel properly

hellosway, 2022-03-14:
If `devicetree` or `ACPI+devicetree` is enabled for `system table selection` vc4 is working.

jlinton, 2022-08-23:
I have a fix for the DT PCIe issue, which shows up on the more recent SoC revisions as well. (basically updating more of the DT with the programming edk2 is doing, rather than expecting it to be able to reset the bridge windows).

jlinton, 2023-05-03:
Well, I don't know what happens WRT DT in that particular case, but this has been a debian specific issue AFAIK because its not symlinking/providing the firmware. AKA the wifi has been working for a couple years with fedora and ACPI (and DT too, although I rarely test it).

And yes this platform provides DMI, you can verify it with `dmidecode` or 'lscpu` or various other utilities.

Also note: the linux-firmware links the DMI string for the wifi firmware on this platform.

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/WHENCE#n2708

jayaddison, 2023-05-03:
Thanks, @jlinton - I wasn't aware of the `WHENCE` file and the `copy-firmware.sh` scripting to create symlinks from it (I had been considering `linux-firmware.git` as basically version control over a set of static files, although now I understand that there are buildsystem components, such as the `Makefile`).

I've likely raised some unnecessary noise within Debian as a result, since I felt that the behaviour of the `brcmfmac` driver (to probe for filenames containing spaces) was odd and out-of-sync with `linux-firmware.git` (apparently not, though).  I'll follow-up there with some explanatory notes and a reference to this discussion.

That said: while I understand that ACPI and Devicetree each have their own heritage, it seems like it could be simpler for most people involved if the requested firmware filename for each device could be the same regardless of the deviceinfo context (apologies: this isn't my area of expertise, quite obviously, so my terminology may be quite inaccurate).

jayaddison, 2023-05-08:
> Thanks, @jlinton - I wasn't aware of the `WHENCE` file and the `copy-firmware.sh` scripting to create symlinks from it (I had been considering `linux-firmware.git` as basically version control over [...]

I've opened a potential fix to handle the spaces-in-filenames problem for the RPi firmware files at: https://salsa.debian.org/kernel-team/firmware-nonfree/-/merge_requests/65 (tracked against Debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1029843)

alsldlflglhljlk, 2023-10-26:
See issue #247 .

alsldlflglhljlk, 2024-03-16:
I think the Wi-Fi problem is solved. I've been able to use Wi-Fi as the main installation method on my latest install.

jayaddison, 2024-08-12:
> * [ ]  Debian installer can't locate WiFi firmware blobs unless the USB is replugged. And it seems to choke on spaces, i.e. should request `brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi 4 Model B.txt` but truncates that to `brcmfmac43455-sdio.Raspberry` instead. Also, why the heck does Debian request `brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi 4 Model B.txt` alongside `brcmfmac43455-sdio.txt`?

There's been some progress here - thanks to [Salsa merge request `firmware-non-free#94`](https://salsa.debian.org/kernel-team/firmware-nonfree/-/merge_requests/94), filenames with spaces are included in the Debian `firmware-brcm80211` package in `trixie`:

```
$ apt-file list firmware-brcm80211 | grep -i raspberry.*4
firmware-brcm80211: /usr/lib/firmware/brcm/brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi 4 Model B.txt
firmware-brcm80211: /usr/lib/firmware/brcm/brcmfmac43455-sdio.Raspberry Pi Foundation-Raspberry Pi Compute Module 4.txt
firmware-brcm80211: /usr/lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.txt
firmware-brcm80211: /usr/lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt
```

As a recap: filenames-with-spaces are requested when the system is configured to use ACPI, and filenames-without-spaces imply that the system is using DeviceTree.

However, there is at least one remaining problem: I doubt that the Debian installer (specifically, the `hw-detect` component) detects this firmware as required during the install process.

I _think_ the fix may be as simple as an edit to a [regex in `check-missing-firmware.sh`](https://salsa.debian.org/installer-team/hw-detect/-/blob/961c536c29f01c47da202034b371caab697d85ac/check-missing-firmware.sh#L157) -- the `firmware: failed to load ([^ ]+)` part in particular won't match filenames with spaces.  But the format of that log message is well-defined in Debian (it originates from [a line in Debian's `linux` package patch `firmware_loader-log-direct-loading-failures-as-info-for-d-i.path`](https://sources.debian.org/src/linux/6.10.3-1/debian/patches/debian/firmware_loader-log-direct-loading-failures-as-info-for-d-i.path/#L32)).  That would require testing, though.

jayaddison, 2024-08-27:
> There's been some progress here - thanks to [Salsa merge request `firmware-non-free#94`](https://salsa.debian.org/kernel-team/firmware-nonfree/-/merge_requests/94), filenames with spaces are [...]

FWIW: it turns out that a fix would not be as simple as that; in particular, spaces are relied upon as delimiter/separator characters elsewhere in the same script, so the addition of filenames containing spaces would cause problems.