pftf/RPi4 #78: Unable to get UEFI setting changes, such as Serial instead of Graphical, to stick if made on 8 GiByte RPi4 (USB-MSB booting)

markmi, 2020-06-24:
In experimenting to see if FreeBSD would boot based on UEFI v1.16's RPI_EFI.fd I via a USB3 SSD GPT partitioned, I did the following sequence. I updated to the stable (not beta) USB-MSB context for both RPi4's. I then tried the 8 GiByte RPi4 first. But the UEFI settings that I attempted to change would not stick. Trying the 4 GiByte RPi4 v1.1 to make those same changes worked fine and the result booted fine. Moving the media back to the 8 GiByte RPi4 used the changed settings and booted just fine. (Each RPi4 had a microsd card in the card slot, with an empty msdosfs, in order to avoid the UEFI code looping indefinitely.)

(I use the same media to boot and operate a Rock64 [not via UEFI], explaining some text below.) For reference for how things ended up looking once booted on the 8 GiByte RPi4:

```
# gpart show -p
=>       40  468862048    da0  GPT  (224G)
         40       2008         - free -  (1.0M)
       2048  413138944  da0p1  freebsd-ufs  (197G)
  413140992    9437184  da0p2  freebsd-swap  (4.5G)
  422578176     204800  da0p3  ms-basic-data  (100M)
  422782976   46079112         - free -  (22G)

# df -m
Filesystem             1M-blocks  Used  Avail Capacity  Mounted on
/dev/gpt/Rock64root       195378 57895 121852    32%    /
devfs                          0     0      0   100%    /dev
/dev/msdosfs/RPI4EFIFS        99     7     92     7%    /usb_efi

# find /usb_efi/ -print | sort
/usb_efi/
/usb_efi/EFI
/usb_efi/EFI/BOOT
/usb_efi/EFI/BOOT/BOOTAA64.EFI
/usb_efi/OVERLAYS
/usb_efi/OVERLAYS/disable-bt.dtbo
/usb_efi/OVERLAYS/miniuart-bt.dtbo
/usb_efi/RPI_EFI.fd
/usb_efi/Readme.md
/usb_efi/bcm2711-rpi-4-b.dtb
/usb_efi/config.txt
/usb_efi/fixup4.dat
/usb_efi/start4.elf

# more /usb_efi/config.txt
arm_64bit=1
enable_uart=1
uart_2ndstage=1
enable_gic=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
disable_overscan=1
device_tree_address=0x1f0000
device_tree_end=0x200000
dtoverlay=disable-bt
over_voltage=6
arm_freq=2000

# vmstat
 procs    memory    page                      disks     faults       cpu
 r  b  w  avm  fre  flt  re  pi  po   fr   sr da0 pa0   in   sy   cs us sy id
 0  0  0 206M 7.6G   18   0   0   0   21    3   0   0   82   21  380  0  0 100
```
(Booting does not mean that there are no operational problems. But it is early for FreeBSD for this area.)

andreiw, 2020-07-11:
Sorry for being a bit slow, but are we saying the bug here is UEFI settings not sticking when:
1) booting UEFI from USB, not SD
     - AND -
2) Pi is 8 GB?

A

markmi, 2020-07-11:
> Sorry for being a bit slow, but are we saying the bug here is UEFI settings not sticking when:
> 
> 1. booting UEFI from USB, not SD
>    
>    * AND -
> 2. Pi is 8 GB?

That is the combination that I was working with that failed. The 4 GiByte RPi4 "booting UEFI from USB, not SD" was able to save setting changes and have them stick.

samerhaj, 2020-08-26:
@markmi can you please confirm if you still see this with the 1.19 FW? There seems to be at least two other issues that sound similar, although not confirmed to be isolated to 8GB: https://github.com/pftf/RPi4/issues/83 and https://github.com/pftf/RPi4/issues/95

markmi, 2020-08-26:
On 2020-Aug-26, at 10:53, Samer El-Haj-Mahmoud <notifications@github.com> wrote:

> @markmi can you please confirm if you still see this with the 1.19 FW? There seems to be at least two other issues that sound similar, although not confirmed to be isolated to 8GB: #83 and #95

Okay. As requested, I've just updated from v1.17 materials
to v1.19 materials on the USB3 SSD that I use for FreeBSD
on RPi4B's.

With a 8 GiByte RPi4B, Graphical -> Serial (for example)
in the UEFI user interface just returned to Graphical, not
actually having saved the updated default (F10).

I still had to use a 4 GiByte RPi4B to do Graphical -> Serial
and have stick after saving (F10). Then going back to using a
8 GiByte RPi4B with the USB3 SSD got the Serial selection
that had been saved via the 4 GiByte RPi4B.

Reminder: No microsd card is involved. This is booting via
the USB3 SSD at all stages. For reference:

# gpart show -pl
=>       40  468862048    da0  GPT  (224G)
         40       2008         - free -  (1.0M)
       2048  413138944  da0p1  RPi4Broot  (197G)
  413140992    9437184  da0p2  RPi4Bswap  (4.5G)
  422578176     204800  da0p3  RPi4BEFI  (100M)
  422782976   46079112         - free -  (22G)

# gpart show -p
=>       40  468862048    da0  GPT  (224G)
         40       2008         - free -  (1.0M)
       2048  413138944  da0p1  freebsd-ufs  (197G)
  413140992    9437184  da0p2  freebsd-swap  (4.5G)
  422578176     204800  da0p3  ms-basic-data  (100M)
  422782976   46079112         - free -  (22G)

# df -m
Filesystem            1M-blocks   Used Avail Capacity  Mounted on
/dev/gpt/RPi4Broot       195378 112924 66824    63%    /
devfs                         0      0     0   100%    /dev
/dev/msdosfs/RPI4BEFI        99     10    89    10%    /usb_efi

-rwxr-xr-x  1 root  wheel    47471 Aug 13 21:05:08 2020 /usb_efi/bcm2711-rpi-4-b.dtb
-rwxr-xr-x  1 root  wheel     5420 Aug 13 21:05:08 2020 /usb_efi/fixup4.dat
-rwxr-xr-x  1 root  wheel  2277856 Aug 13 21:05:08 2020 /usb_efi/start4.elf
-rwxr-xr-x  1 root  wheel     5716 Aug 13 21:02:28 2020 /usb_efi/Readme.md
-rwxr-xr-x  1 root  wheel      206 Aug 13 21:02:28 2020 /usb_efi/config.txt.uefi_orig
-rwxr-xr-x  1 root  wheel  2031616 Aug 13 14:10:06 2020 /usb_efi/RPI_EFI.fd
-rwxr-xr-x  1 root  wheel      236 Jul 15 02:21:48 2020 /usb_efi/config.txt

-rwxr-xr-x  1 root  wheel  1819 Aug 13 21:05:08 2020 /usb_efi/OVERLAYS/miniuart-bt.dtbo
-rwxr-xr-x  1 root  wheel  1073 Jul 14 18:43:14 2020 /usb_efi/OVERLAYS/disable-bt.dtbo

===
Mark Millard
marklmi at yahoo.com
( dsl-only.net went
away in early 2018-Mar)

markmi, 2020-09-09:
On 2020-Aug-26, at 11:55, Mark Millard <marklmi26-fbsd at yahoo.com> wrote:

> On 2020-Aug-26, at 10:53, Samer El-Haj-Mahmoud <notifications@github.com> wrote:
>
>
>
> Okay. As requested, I've just updated from v1.17 materials
> to v1.19 materials on the USB3 SSD that I use for [...]

FYI: I updated to v1.20 (from v1.19) yesterday on the
8 GiB RPi4B's in the USB3-only (no microsd card) context.
I changed Graphical to Serial and saved and it via F10
and the result stuck. I did not have to use a 4 GiB model
to save the changes.

(Note: I do this change in part because FreeBSD -r363590
still crashes early in its boot sequence when Graphical
is selected.)

===
Mark Millard
marklmi at yahoo.com
( dsl-only.net went
away in early 2018-Mar)

markmi, 2020-09-10:
On 2020-Sep-8, at 17:33, Mark Millard <marklmi26-fbsd at yahoo.com> wrote:
> FYI: I updated to v1.20 (from v1.19) yesterday on the
> 8 GiB RPi4B's in the USB3-only (no microsd card) context.
> I changed Graphical to Serial and saved and it via F10
> and the result stuck. I did not [...]

I tried another v1.20 install today and this time the F10
based saves did not stick on the 8 GiB (multiple tries)
but did stick when tried from the 4 GiB (first try, as
always so far).

This indicates that the problem on the 8 GiB is
intermittent. (But the activity is not one I normally
do frequently so the variability is not well classified.
It also means that testing needs a bunch of successes
to strongly hint overall success.)

===
Mark Millard
marklmi at yahoo.com
( dsl-only.net went
away in early 2018-Mar)

oldsadsongs, 2020-11-23:
Can confirm that this is still an issue with v1.21 - 8gb with a usb drive only.

oldsadsongs, 2020-11-26:
Posting the work-around that I found on discord here - Partition and format a SD card, unzip, boot and save your changes. Copy your EFI partition from your SD card to your USB drive.

hille721, 2020-12-05:
I had the same issues with v1.21 and 8GB only using a SSD.
Workaround from @oldsadsongs above works well.

pbatard, 2020-12-19:
Soooo, after much of adding debug statements from `VarBlockServicesDxe` down to `FatPkgDxe`, it turns out that this is a simple issue of write caching with USB mass storage devices, especially SSD based ones.

In other words, unless it's only referencing a write cache that has an explicit USB Mass Storage command for flushing, it doesn't look like [`This comment from UsbMassImpl.c`](https://github.com/tianocore/edk2/blob/6932f4bfe552c1704c5715430de6045c78a5b62f/MdeModulePkg/Bus/Usb/UsbMassStorageDxe/UsbMassImpl.c#L308) is correct, because it's easy to demonstrate that some USB Mass Storage devices (I have at least 2 of those, and I haven't really looked that hard) **do** implement write caching.

For instance, one can insert an ASSERT at the end of `VarBlockServicesDxe.c`'s `DoDump ()` to control platform reset manually, and if you do that, with an SSD based USB 3.0 device (and probably non SSD based flash drives too), you will find that:
* If you wait 3-5 seconds, everything's fine and reading back the data after unplugging the drive produces the expected NV alterations
* Wait less than 1 second data and NV data is missing on the file system because it wasn't flushed from the write cache.

And since we're stalled on an ASSERT, you can't pin that behaviour on anything else but the Mass Storage device itself, even more so as we do all the right things in our NV Dump with [such as issuing `Flush()` before `Close()`](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Drivers/VarBlockServiceDxe/FileIo.c#L33-L40). I even went as far as re-opening the file and re-reading the data that just got written, after we close it in `DoDump()`, to validate that, as far as UEFI is concerned, we're getting a clear message from the Mass Storage device that, yes, our data has been written, even as it's clearly only residing in the internal device write cache and not yet on the flash itself.

As to why we're not seeing this with SD, well, of course, SD doesn't have that issue, since SD speeds are much slower than USB 3.0, and therefore, the need for async/cached internal operations is much smaller.

Which means we have two approaches to address this:
1. Go through a wild goose chase in edk2's `UsbMassStorageDxe` to try to figure out if it's possible to control Mass Storage write caching (Microsoft seems to indicate that this **may** be possible in the policy you can apply to a USB Mass Storage device in Windows with their _"Disables write caching **on the device** and in Windows (...)"_) or implement some kind of actual flush that flushes the drive's internal cache.  
[image]

2. Add a few seconds sleep delay before resetting the platform (NB: We don't want to do that in `DoDump()` because it can and __is__ callled multiple times in standard operations) to give time for Mass Storage media that have a write cache to flush their data.

Obviously, I have a strong preference for option 2, because this is a very vexing issue, that has been plaguing people for way too long, and I think we will better serve Pi UEFI users by fixing this ASAP rather than agonizing on corner case scenarios (_"But what if some guy has a device where n seconds are not enough to flush the write cache?"_) that are unlikely to benefit much of anyone, even more so if we have to invest tons of resources to solve them.

As such, I am planning to send a patch that adds something like a 3 or 5 seconds delay in [`ResetLib`](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Library/ResetLib/ResetLib.c#L54) if we have dumped the NV variables, since I tested that doing so unconditionally appears to solve the issue on my platform. But I'm open to other suggestions if you have any...

markmi, 2020-12-19:
I'm surprised at the explanation, given that I've never seen the problem when using the same USB SSD device on the 4 GiByte RPI4's, only on the 8 GiByte RPi4's: Same USB device characteristics since it is physically the same device. So I would have expected some 8 GiByte-RPi4-context-specific involvement in the explanation. But it may fit with having had it work on the 8 GiByte RPi4's on rare occasion. (I had another example recently.) I normally use a 4 GiByte RPi4 to do the setting changes.

Still, it certainly seems worth seeing if the extra delay makes updating settings more reliable on the 8 GiByte RPi4's in my context, even if there is no explanation for the 4 GiByte RPi4's differing behavior.

markmi, 2020-12-19:
Does the File->Flush (File) end up invoking UsbMassFlushBlocks --that in turn is a no-op but for the constant return value? If yes, then the File->Flush (File) in Close may not be doing as intended. (Be warned that I'm outside my knowledge base for this note. Sorry if it is just noise.)

pbatard, 2020-12-19:
My best guess for 4 GB vs 8 GB difference is that the platforms may have different times between reset/shutdown init and actual reset/shutdown, which the mass storage device can use to flush it's write cache. Or it could be that there is a way to disable write caching for USB Mass Storage devices, which the **closed source** Raspberry Pi firmware (`start4.elf`), that's invoked before the  UEFI firmware, may set by default on 4 GB but not on 8 GB...

I am however positive it can't be anything else but write caching at this stage, as I demonstrated that you can halt CPU execution altogether (meaning, there is no more code executing, at least on the CPU side, that could even remotely interfere with the write operation or somehow rewrite old data &mdash; something I also envisioned), and the data will or will not be written depending on how much time you let elapse before you manually shut down the platform after the CPU is stalled. In other words, the difference of behaviour can't be tied to anything executing CPU-side.

As to `Flush()` not doing what one might expect, that's precisely the point I made with option 1, which is the one option I want to avoid (unless **somebody else** is willing to invest their own time figuring out how one can implement a proper Flush for Mass Storage media that have internal write cache).

markmi, 2020-12-19:
When I edit and save settings, I then continue in the UEFI menus until I reach the outer level and finally cause the restart. I may or may not adjust more settings and save again along the way. Sometimes I'm directly going to the outer level. I've never noticed any pattern relative to such variations. But I expect that it takes some notable time before I cause an explicit restart. If there are not other internal USB resets and sleep-time ends up as a fix, then the time likely needs to be longer than such activities were? May be this provides a way to explore how long to try in the sleep?

pbatard, 2020-12-19:
The dumping of the user modified variables is actually an event that's registered for when UEFI reset/shutdown occurs (these changes are **not** saved as you make them).

So the current process is pretty much as follows:
1. User does whatever in about as much time as they want, and once they are done, they initiate an operation to either reset or shut the platform down.
2. UEFI receives the request for Reset/Shutdown
3. UEFI sends a custom event that has been previously registered by VarBlockServicesDxe to be informed of when the system is about to shutdown/reset
4. VarBlockServicesDxe checks if any variables have been altered and need to be dumped. If so it saves them
5. VarBlockServicesDxe hands back over to UEFI
6. UEFI resets/shuts down the platform

So, as a user, you have no way to influence the time that occurs between the writing/dumping of the variables and reset. And, because of standard UEFI design, these two things are actually set to happen as fast as possible one after the other, since they are actually tied, which is the source of our issue, because it gives no time for Mass Storage devices that may still have data in their write cache to empty it.

Especially, please don't be fooled into thinking that when you see the message that asks you _Settings have been altered. Do you want to save them?_, then the variables are being dumped immediately if you press `Y`. The dump only happens when the platform is actually about to reset.

samerhaj, 2020-12-20:
The analysis (and option 2) seems reasonable as a quick workaround for this issue.

But it seems this would not explain the Async Exception seen in #97 (and duplicate issues) which is linked to variable storage on SD card. @pbatard do you think the "not saving settings on USB" issue described here is different from the Async Exception suspected variable store corruption issues?

pbatard, 2020-12-20:
Well, I could see SD NV corruption as a result of SD write cache being in the process of being flushed when reset occurs. Whereas for USB, the reset occurs before write cache is being flushed, resulting in a variable store that has not been updated at all, for SD it may occur while the SD card is in the process of updating the file, resulting in a variable store that is corrupted.

At any rate, I think the proper way to look at it is not to try to speculate but see what happens once we have applied the patch that adds a delay before reset on NV update.

samerhaj, 2020-12-20:
I can test with your patch if you have it somewhere (using the procedure in #95).

pbatard, 2020-12-21:
Don't have anything I can provide yet. I'll make sure to CC you when I send it to the list.

pbatard, 2020-12-22:
Patch has now been sent to the list in https://edk2.groups.io/g/devel/message/69384.

pbatard, 2021-01-06:
This should be fixed in UEFI firmware v1.22. If that is not the case, please let us know.

closed 2021-01-06 by pbatard

mistermatt2u, 2021-01-16:
@pbatard I think there might still be a problem. On a RPi 4GB, when I disable the PXE and HTTP boot stuff, it still shows up on the list after a save and reboot. More importantly, it looks like the iSCSI menu disappeared completely...

pbatard, 2021-01-16:
> More importantly, it looks like the iSCSI menu disappeared completely...

Duplicate of #125, which has nothing to do with this change.

With regards to your other issue, it's also unlikely to be related to this one __unless__ you also don't see changes like > 3 GB RAM or Graphical vs Serial console selection for default console being persisted. Do you see an issue with these changes being saved too? And are you asked to save your changes after you change your PXE/HTTP options?

mistermatt2u, 2021-01-17:
Sorry, I'm new here. Didn't realize the iSCSI issue was already being tracked. Thanks for that.

The ">3 GB RAM" change is saving ok. I am indeed asked to save after deleting the PXE/HTTP options. They just don't actually delete. They seem to be moved to the bottom of the boot order.... but they are still attempted.

samerhaj, 2021-01-17:
That Is an expected behavior for the PXE boot options. Even if you delete them, they will get rediscovered automatically by the UEFI FW on next boot, and get added to the bottom of the list.

mistermatt2u, 2021-01-17:
@samerhaj Thanks for the clarification. I believe that is what is occurring. However, on one RPi 4GB device, using 1.20, the boot order kept prioritizing PXE/HTTP over iSCSI. Those failed in a way such that iSCSI never got attempted. Thats what drove me to update to 1.22...

I might give 1.21 a go. But if iSCSI isn't really recommended anymore, I will probably just go with USB boot storage.

samerhaj, 2021-01-17:
The disappearance of iSCSI support in the latest FW is a bug. Hopefully, it will be fixed in the next release.

mateuszdrab, 2021-02-05:
I just installed the 1.22 version of EFI on my RPI where I'm USB booting and as soon as I enabled 8GB RAM, it stopped booting from USB and now goes through all PXE options and then states no device was found. It seems that the boot override/order options are ignored and I have to manually launch the boot menu to select the USB device to boot.

commit d8a55b01 (2021-01-06): Update to EDK2 latest

* Closes #78
* Closes #82
* Closes #97