pftf/RPi4 #80: Items requiring solving for Windows 10 support

pbatard, 2020-07-09:
Here are all the items that require solving, for the Windows option to become somewhat usable on the Pi 4:

### Major showstoppers

- [ ] `usbxhci.sys` needs to use double DWORD access to read/write 64-bit xHCI registers, instead of QWORD. &rarr; Without this, the VL805 xHCI controller doesn't work and none of the rear-panel USB-A ports can be used.
- [ ] UEFI boot should not freeze when no SD card is present. &rarr; Classified as major showstopper as it is a lot more convenient to create and boot a single Windows USB boot media, than have to require a separate SD. Tracked in #37.
- [ ] Windows 10 ARM64 Genet driver &rarr; Any OS is next to useless without native Ethernet connectivity, and requiring an external USB Ethernet adapter (for which few manufacturers seem to have Windows 10 ARM64 drivers) is less than convenient. Considering how unlikely it is that Broadcom will be investing resources to that effort, the Genet driver __will__ have to come from a third party...

### Partial showstoppers

- [ ] 3 GB RAM limit &rarr; When booting from rear USB, we can use up to 3 GB RAM (as opposed to 1 GB when booting from USB-C), but it'd be a lot nicer to have the full 4 or 8 GB RAM available, for those who have these models.
- [ ] Wifi driver &rarr; Same story as Genet. However, if we can solve wired connectivity, this isn't as critical.
- [ ] SD driver

### Minor inconveniences

- [ ] Audio (from jack) is broken. &rarr; Most likely because the Pi 3 driver being reused does not apply to Pi 4...
- [ ] PL011 serial works, but the baudrate needs to be set to half the desired value. &rarr; We might be able to "patch" the existing Pi 3 driver to fix that...

Biswa96, 2020-07-09:
> Windows 10 ARM64 Genet driver → Any OS is next to useless without native Ethernet connectivity, and requiring an external USB Ethernet adapter

Maybe one can use Android device for internet. Though I have not tried it with Windows 10 but I have used my Android phone for internet with USB tethering.

Biswa96, 2020-07-09:
> Windows 10 ARM64 Genet driver → Any OS is next to useless without native Ethernet connectivity, and requiring an external USB Ethernet adapter

Maybe one can use Android device for internet. Though I have not tried it with Windows 10 but I have used my Android phone for internet with USB tethering in Debian.

pbatard, 2020-07-09:
Was trying to delete the double comment, but looks like @Biswa96 was doing the same, and picked the other one.

I wasn't trying to censor thing, so here's what was posted:

>> Windows 10 ARM64 Genet driver → Any OS is next to useless without native Ethernet connectivity, and requiring an external USB Ethernet adapter

> Maybe one can use Android device for internet. Though I have not tried it with Windows 10 but I have used my Android phone for internet with USB tethering in Debian.

binarymaster, 2020-12-09:
> and what about win10arm32?

@bzy-080408 for this you will need 32-bit UEFI firmware, which I believe is out of this project scope.

But pull requests welcome as always, I think. :slightly_smiling_face:

bzy-080408, 2020-12-11:
@binarymaster I just run windows RT8.1 on it use this uefi,but it's not start.So i think it can't boot the arm32 system.

CoiaPrant, 2021-09-21:
Is it possible to adapt the GPU driver and HDMI audio driver?

paulwratt, 2022-02-17:
I believe the WiFi Issue from this List is the same problem as the Debian List. Hasn't a solution to this already been found by WoR (or WoA)?

The Genet driver has been updated, is it not possible to verify those changes against the WiFi isssue?

Whats the status of the 2x Minor Inconveniences?