pftf/RPi4 #82: Resetting with CTRL+ALT+DEL

samerhaj, 2020-07-18:
Resetting the system with CTRL+ALT+DEL does not trigger non-volatile variables flush to disk. Tested by changing some settings in Setup UI and saving (F10) then CTRL+ALT+DEL. The same works if resetting using the "Reset System" menu

samerhaj, 2020-12-21:
The UsbKbDxe driver [traps the CTRL+ALT+DEL and sends a gRT->SystemReset() call,](https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Bus/Usb/UsbKbDxe/KeyBoard.c#L1470) which should trigger with the [RPi ResetLib ](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Library/ResetLib/ResetLib.c#L52), and ends up calling [DumpVars](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Drivers/VarBlockServiceDxe/VarBlockServiceDxe.c#L151).  So this does seem to be correct.

It may be that #78 is involved

pbatard, 2021-01-06:
This should be fixed in UEFI firmware v1.22. If that is not the case, please let us know.

commit d8a55b01 (2021-01-06): Update to EDK2 latest

* Closes #78
* Closes #82
* Closes #97
closed 2021-01-06 by pbatard