pftf/RPi4 #88: SCT Failure - Usb2Hc  protocol API failures

samerhaj, 2020-07-27:
From: https://github.com/pftf/acs-reports/blob/master/v1.17%2B/sct-tianocore/Overall/Summary.log

Running Full SCT test on latest RPi code,

```
Reset - Call Usb2Hc->Reset() with Invalid Parameters -- FAILURE
D2E6A8F0-6C97-4134-812E-25F175186AE4
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:1874:Status - Success

Returned Status Code: Success

Reset_Conf: [FAILED]
  Passes........... 0
  Warnings......... 0
  Errors........... 1

USB2_HC_PROTOCOL.SetState - SetState() returns EFI_INVALID_PARAMETER with Invalid Parameters -- FAILURE
5168C4EF-91F4-48C5-881F-F80180D29807
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:2022:Status - Success

USB2_HC_PROTOCOL.SetState - SetState() returns EFI_INVALID_PARAMETER with Invalid Parameters -- FAILURE
5168C4EF-91F4-48C5-881F-F80180D29807
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:2046:Status - Success

Returned Status Code: Success

SetState_Conf: [FAILED]
  Passes........... 0
  Warnings......... 0
  Errors........... 2

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with Data is NULL -- PASS
3A0AD565-B82C-450F-BCE6-88B3D16ADE35
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:3987:Status - Invalid Parameter

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with DataLength being 0 -- PASS
C139127A-3797-482F-B35C-AAF799BDF6C6
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4079:Status - Invalid Parameter

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with MaxiumPacketLenth is not valid -- FAILURE
14CB206C-422B-47EE-8C4B-F316FE33DAFB
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4174:Status - Device Error

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with MaxiumPacketLenth is not valid -- FAILURE
14CB206C-422B-47EE-8C4B-F316FE33DAFB
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4209:Status - Device Error

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with MaxiumPacketLenth is not valid -- FAILURE
14CB206C-422B-47EE-8C4B-F316FE33DAFB
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4252:Status - Device Error

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with DataToggle points to a value other than 0 and 1 -- PASS
F4353439-47E4-4DF3-85E9-9EFE723A1E4B
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4344:Status - Invalid Parameter

USB2_HC_PROCOTOL.SyncInterruptTransfer - SyncInterruptTransfer() returns EFI_INVALID_PARAMETER with TransferResult is NULL -- PASS
81DFDB23-681E-4DF7-A773-6D4158DB883E
/mnt/c/dev/tianocore/SctPkg/TestCase/UEFI/EFI/Protocol/Usb2Hc/BlackBoxTest/Usb2HcTestConformance.c:4436:Status - Invalid Parameter

Returned Status Code: Success

SyncInterruptTransfer_Conf: [FAILED]
  Passes........... 4
  Warnings......... 0
  Errors........... 3
```

samerhaj, 2020-12-24:
patch sent to EDK2 list: https://edk2.groups.io/g/devel/topic/edk2_platform_patch_v1_1_1/79208713

ljfranklin, 2021-03-07:
@samerhaj I'm booting the [iPXE EFI](https://github.com/ipxe/pipxe) application on the Raspberry Pi with EDK2 and it looks like [the commit](https://github.com/tianocore/edk2-platforms/commit/f8958b86e8863432b815a132a0f0fe82950c6dd1#diff-a4c80030ebbe12ded482507369d24076dbca1e8c3a42d0422f63958f2c93f86f) attached to this issue is causing iPXE to fail to initialize the Pi's network devices. If I build EDK2 with that commit or later, iPXE reports "No more network devices" while if I revert that commit or use an earlier commit the network devices load as expected.

To reproduce the issue you can burn the [iPXE RPI image](https://github.com/ipxe/pipxe/releases/tag/v20191103) to an SD card and replace the `RPI_EFI.fd` file with one built from latest EDK2 and verify the network devices fail to load.

samerhaj, 2021-03-09:
Interesting.. Thanks for reporting this. That commit only changed the USB drivers/stack to be more compliant with the UEFI specification and handle some error conditions. Can you please open a new Github issue to track this problem ? By any chance, do you have USB devices (such as a NIC) that are in use? You may also want to collect the UEFI serial debug log and attach it to the issue. Please let me know if you need instructions for that.

samerhaj, 2021-03-11:
@ljfranklin  could you please look at: #122 (comment) , and try that FW binary or change to see if it fixes the issue with iPXE? Are you using a USB NIC adapter by any chance?

ljfranklin, 2021-03-12:
Responded in the thread you linked. Does seem like this patch fixes the issue I was seeing: https://github.com/rtreffer/edk2-platforms/commit/ca053c6. Thanks for taking a look!

samerhaj, 2021-03-12:
Fix committed upstream: https://github.com/tianocore/edk2-platforms/commit/2620e05c6fade20ef5e1ba69280d09f9e2b3eff6

closed 2021-07-16 by samerhaj