pftf/RPi4 #93: ACS FWTS uefirttime - RT services failure with SetTime()

samerhaj, 2020-08-26:
From the new ACS 2.5 beta tests on Rel 1.19:

https://github.com/pftf/acs-reports/blob/master/v1.19/acs-enterprise-v2.5-beta/luv-results-2020-08-24--10-11-06/raw/fwts#L223

```
1.1  fwts uefirttime: INFO UEFI Runtime service time interface tests.
...
1.1  fwts Test INFO 4 of 36: Test UEFI RT service set time interface.
1.1  fwts FAILED INFO [HIGH] UEFIRuntimeSetTime: Test 4, Failed to set time with UEFI runtime
1.1  fwts service. INFO service.
1.1  fwts Return INFO status: EFI_OUT_OF_RESOURCES. A resource has run out.
...
1.1  fwts Test INFO 5 of 36: Test UEFI RT service set time interface, invalid year 1899.
1.1  fwts PASSED: INFO Test 5, UEFI runtime service SetTime interface test passed, returned
1.1  fwts EFI_INVALID_PARAMETER INFO as expected.
1.1  fwts FAILED INFO [HIGH] UEFIRuntimeSetTime: Test 5, Failed to set wakeup time with UEFI
1.1  fwts runtime INFO service.
1.1  fwts Return INFO status: EFI_DEVICE_ERROR. The physical device reported an error while
1.1  fwts attempting INFO the operation.

...
1.1  fwts 15 INFO passed, 2 failed, 0 warning, 0 aborted, 20 skipped, 0 info only.
1.1  fwts 
```

samerhaj, 2020-11-16:
Issue seems to be related to storing the TZ and Daylight portion of EFI_TIME. The EmbeddedPkg VirtualRealTimeClock driver calls SetVariable() to store those settings in NV variable store, which results in an error returned to the SetTime function, that gets passed back to the caller: https://github.com/tianocore/edk2/blob/master/EmbeddedPkg/Library/VirtualRealTimeClockLib/VirtualRealTimeClockLib.c#L291

A better way to handle this is for the UEFI SetVariable service to return UNSUPPORTED (per https://github.com/pftf/RPi4/issues/112), and for the VirtualRealTimeClock code to detect that, and return gracefully to the caller without failure. Of course TZ/DL settings will not be stored in NV, but none of the time is stored in an RTC / retained across reboots anyway.

samerhaj, 2021-07-16:
Also Note that FWTS checks for EFI_RT_PROPERTIES table, and skips these tests if the table properly flags the service as un-implemented (seee https://git.launchpad.net/fwts/tree/src/lib/src/fwts_uefi.c#n553). However, the RPi (and EDK2 in general) does not implement this table yet, as reported in #112 . Once that is implemented, these FWTS failures will get resolved

samerhaj, 2021-07-16:
Also linking to #6 . If we have a write-able NV store at runtime, this issue will be resolved

sunnywang-arm, 2021-07-23:
Sent a fix to the edk2 mailing list - https://edk2.groups.io/g/devel/topic/edk2_platform_patch_v1_1_1/84397460

sunnywang-arm, 2021-07-29:
> Sent a fix to the edk2 mailing list - https://edk2.groups.io/g/devel/topic/edk2_platform_patch_v1_1_1/84397460

I dropped the fix above because it would cause OS installation failure/error. 
The root cause of this issue is the same as #163.  I also proposed a solution and added more information in #163