pftf/RPi4 #94: ACS FWTS ACPI table revision check

samerhaj, 2020-08-26:
From the new ACS 2.5 beta tests on Rel 1.19:

https://github.com/pftf/acs-reports/blob/master/v1.19/acs-enterprise-v2.5-beta/luv-results-2020-08-24--10-11-06/raw/fwts#L1741

```
1.1  fwts Test INFO 2 of 2: Test ACPI spec versus table revisions.
1.1  fwts System INFO supports ACPI 0630
1.1  fwts Table INFO APIC has a matched revision.
1.1  fwts Table INFO BGRT has a matched revision.
1.1  fwts FAILED INFO [MEDIUM] ACPITableBadRevision: Test 2, ACPI Table DSDT revision was
1.1  fwts expected INFO to be 2, got 5.
1.1  fwts Table INFO GTDT has a matched revision.
1.1  fwts Table INFO RSDT has a matched revision.
1.1  fwts FAILED INFO [MEDIUM] ACPITableBadRevision: Test 2, ACPI Table XSDT revision was
1.1  fwts expected INFO to be 1, got 2.
```

samerhaj, 2021-04-16:
There is a statement from the UEFI Forum on this: https://uefi.org/node/4185

There has also been some patches sent to EDK2 to correct ACPI table versions for other platforms: https://edk2.groups.io/g/devel/message/73981?p=,,,20,0,0,0::Created,,update+ACPI+table+revision,20,2,0,82046081

We should do the same for the RPi4 code

sunnywang-arm, 2021-04-26:
> 1.1  fwts FAILED INFO [MEDIUM] ACPITableBadRevision: Test 2, ACPI Table XSDT revision was
1.1  fwts expected INFO to be 1, got 2.

The XSDT revision failure above is an FWTS issue that should already be fixed by the commit https://kernel.ubuntu.com/git/hwe/fwts.git/commit/?id=c522bfedc9839a474b8d590ba36bec77436d2e90 .  ACS is using FWTS V20.08.00 that doesn't include this fix. I will offline check this with the ACS owner.  

> 1.1  fwts FAILED INFO [MEDIUM] ACPITableBadRevision: Test 2, ACPI Table DSDT revision was
1.1  fwts expected INFO to be 2, got 5.

For the DSDT failure,  I will send a patch to the edk2 mailing list for this.

closed 2021-05-10 by ardbiesheuvel