pftf/RPi4 #97: Synchronous Exception at 0x00000000371013D8

TheMindVirus, 2020-09-06:
After a routine restart of the system I get **Synchronous Exception at 0x00000000371013D8** at UEFI boot after the timeout.

[image]

I am able to access the UEFI Boot Menu with the Keyboard and also from the Serial Console (arrow keys and selection).
I have had the same issue intermittently before, both times it prevented boot into Windows 10.
The SD card has been checked and verified, both partitions are healthy and the files seem untouched.
I will try replacing the boot files with new copies next but this is still a recurring issue and is a big killjoy for the user.

TheMindVirus, 2020-09-06:
Update: I have been able to repair my image with a hybrid boot partition from its broken state and a newly flashed boot partition. However, neither the broken one or the newly flashed one worked properly. When I used the EFI folder and custom config from the broken partition with bootcode and fixups from the new partition, the repaired image started working again. This hints at some corruption being caused within one of the boot files, affecting any of the other files in that partition.
I originally suspected some EEPROM corruption but this was not the case.

pbatard, 2020-09-06:
If you encounter an exception, the first thing you should do is ensure that your GPIO is plugged to a serial port and then replace your firmware with the DEBUG version, which you can always find in the [AppVeyor artifacts](https://ci.appveyor.com/project/pbatard/RPi4/build/artifacts).

Then you should post the log you get from serial.

It does seem, from your update, that the exception comes from the bootloader rather than the UEFI firmware, in which case I don't think this is something we need to look into.

TheMindVirus, 2020-09-06:
I recreated the conditions of the broken image on the SD card but using the RPI_EFI.fd DEBUG version from appveyor and some tweaks to get it to boot successfully.
It did indeed boot successfully and here was the Serial Output at 115200 8-N-1.
[SerialOutput.txt](https://github.com/pftf/RPi4/files/5179736/SerialOutput.txt)
As you can probably tell, the procedure happens so fast and somewhat in parallel so some of the data dumps get clobbered by each other. The initial readout from bootcode.bin on the EEPROM looks fine (this is actually what the UART might be coming up against at boot).

pbatard, 2020-09-06:
Something looks very wrong with your setup.

There's nothing going on in parallel in the UEFI firmware, it's all sequential, so, if you properly set your serial connection, you shouldn't be getting anything as garbled as that.

It looks to me like your baudrate is not set properly on the receiver, as, right from the get go, we're seeing invalid characters in your output such as in the one that is produced by the native Raspberry Pi EEPROM:
```
Read fixup4db.dat bytes     8427 hnd 0x00000003 hash 9g2ff1118712591b
```
Obviously the hash should be an hexadecimal number, so having a `g` there means that your baudrate is off. And we're seeing further evidence of this with `UEFH firmware` instead of `UEFI firmware` and then I suppose that when your receiver starts to receive a lot of output in a short time (which the bulk of the DEBUG firmware execution will produce), it completely loses it.

You're going to need to fix your serial input one way or another, as I'm afraid that garbled serial data doesn't help much...

TheMindVirus, 2020-09-06:
Hmm...I'm going to use another Raspberry Pi for this purpose but in this case the output will become biased towards working for Pi and not other platforms. The Serial connection has been tested with many other boards and none of them produce anything like the output as shown previously.

TheMindVirus, 2020-09-06:
Here is the output from a Pi3 to a Pi4 (it was a pain to set up).
[SerialOutput3.txt](https://github.com/pftf/RPi4/files/5179936/SerialOutput3.txt)
It seems to be breaking some sort of speed limit. Whether it is going above 115200 baud or processing close to the buffering limit of the Windows serial relays is unclear, though I'd suggest it's the first because slow output displays correctly and fast output ends up in letters being augmented.

pbatard, 2020-09-06:
For the record, I'm using putty for serial input on Windows, and have never experienced buffering issues, so I don't think it's a Windows issue.

Also, I'm not seeing much out of the ordinary from your log. `FSOpen: Open '\EFI\BOOT\BOOTAA64.EFI' Success` means that the UEFI firmware has reached the hand over point to your boot loader, so, unless your exception occurs before that, the problem is most likely in the bootloader itself (as your testing seems to point out too), in which case I'm not sure why you are reporting the issue here. Anybody can write a UEFI application that will produce a synchronous exception. That does not mean that the UEFI firmware has an issue.

Unless you can provide clear evidence that the Pi 4 UEFI firmware appears to be doing something wrong, I will just have to close this issue as "out of scope" of what we support.

TheMindVirus, 2020-09-06:
It is difficult to pinpoint without knowing exactly what set of instructions get sent to 0x00000000371013D8, such is the security of things like this. The last log sent was one I spent time repairing so that it did boot and something out of the ordinary it did point out was that my MBR had apparently been flashed badly or had errors. Any failure to boot at this stage hints at addressing relative to MBR instead of addressing relative to data read from the discovered files in the partition. Now that the WoR imager supports GPT and new firmware has been released it seems less of an issue, but I'll let you know if I see this error cropping up again.

pbatard, 2020-09-06:
Okay. Make sure you try with the DEBUG firmware and provide a serial log then. I will close this issue for now.

andreiw, 2020-09-08:
I believe this is actually a crash in UEFI caused by a corruption in the "variables FV". I've seen it, and so has @samerhaj. It seems like a previous boot of UEFI caused a corrupted NVRAM section to be flushed to disk.

samerhaj, 2020-09-11:
Yes I have seen an intermittent hang that results in corruption to the NV variable store. Was able to reproduce with SCT: https://github.com/pftf/RPi4/issues/95

Once in that state, any NV variable sets fail to commit to NVRAM unless you delete the RPI_EFI.FD and start over.

Initially I saw this only on the 8GB so I thought it is linked to 8GB specifically (as also reported here https://github.com/pftf/RPi4/issues/78 and here https://github.com/pftf/RPi4/issues/83 )

TheMindVirus, 2020-09-11:
Mine is a Pi4 4GB model. I also saw the issue following the SD card. With two almost identical images, the error was only on one of them and was intermittent.

However, I noticed despite following the normal "sudo apt-get full-upgrade" procedure, my EEPROM had held the version from January 2020.

I then manually upgraded to the most recently available April 2020 EEPROM  version and found out that the package maintainers had one from September 2020 (and had removed EEPROM updating from full-upgrade).

Since I have been on a new image, new SD card, September 2020 EEPROM and v1.20 UEFI, it seems to be running fine. I'll let you know if I see it halting again.

andreiw, 2020-09-12:
I confirm a crash on 1.20 on 8gb model. Right after I moved it from a 4gb model. It appears more likely to happen on 8gb.

TheMindVirus, 2020-09-12:
If you have Raspberry Pi OS to hand you can check the EEPROM Version with "sudo vcgencmd bootloader_version". What is the version of your 4GB model vs the 8GB model?

andreiw, 2020-10-01:
Reopening as this is definitely a problem that manifests itself periodically.

andreiw, 2020-10-01:
Will check-out the hint from @TheMindVirus on bootloader_version. Interesting.

TheMindVirus, 2020-10-05:
I've begun work on a preliminary GNU Makefile to build UEFI without appveyor.
https://github.com/TheMindVirus/RPi4/blob/master/Makefile

Once it's done it will require a fresh Ubuntu 20.04 image and for the dev to type "make".
It could be made into a Docker image afterwards so that it becomes more portable.
The thought had crossed my mind that Ubuntu running on the Pi could built it as well.

I have some changes to the GPIO base address I would like to try out,
so I could hopefully begin that if this starts working.

pbatard, 2020-10-05:
@TheMindVirus, I appreciate the effort, but AppVeyor is certainly __NOT__ required in the slightest to build the firmware, and the default EDK2 scripts (which are based on GNU makefile) will happily build the firmware for you, on any platform that has the relevant toolchain.

We are __only__ using AppVeyor in this specific repository to build zip archives for upload. But that is __NOT__ the official repository for the firmware: https://github.com/tianocore/edk2-platforms/ is, and this repository already has all the elements you need to build the firmware using GNU Make without any reliance on AppVeyor.

I can actually give you the complete script I use to build the firmware on Linux, which stands in a couple of lines. If you installed the prerequisites, it shouldn't be more complicated than that.

andreiw, 2020-10-16:
So it's still definitely a problem, the EEPROM version doesn't matter (actually - whew, that would have been weird if it did)

andreiw, 2020-10-16:
Like @pbatard said, you can manually build it in easy steps.

I generally use https://github.com/andreiw/lampone, which does a bit more (easily rebasing my work, for example)

andreiw, 2020-10-16:
Anyway, I'll assign myself this one. Hopefully it reproes with a DEBUG build.

samerhaj, 2020-11-16:
Do we know if this issue can still be reproduced with the new v1.21 https://github.com/pftf/RPi4/releases/tag/v1.21 (that fixes DT corruption #107) ?

TheMindVirus, 2020-11-16:
The only time I’ve seen it crop up is when trying to enable Virtual Machine Platform from within Windows (which causes an infinite hang on boot on a separate note). Quite likely there is a bug in the installer for the feature in WoA which has not been patched yet.

It occurred because the VMP installer tried to change boot settings on both BCD and in system32/winload.efi (repaired with “bcdboot C:\Windows /s B:”).

It occurs whenever the boot process contains errors, files are missing or if the NVRAM is corrupt (afaik).

samerhaj, 2020-12-20:
Possible root cause and proposed fix in #78

TheMindVirus, 2020-12-20:
I certainly remember having to call sync() a lot from Linux, is that why? Even when switching out the SD card and switching it back again, as mentioned previously the error follows the SD card until it's re-flashed. Definitely does sound like a critical pending write that never happens from what we're all seeing. A call to whatever the UEFI equivalent of sync() is - placed just after the culprit pending write - will more than likely fix it (as it does when you re-flash).

pbatard, 2021-01-06:
This is expected to be fixed in UEFI firmware v1.22. If that is not the case, please let us know.

closed 2021-01-06 by pbatard

jsbackus, 2021-05-31:
Hi folks,

Quick follow-up for documentation purposes: I ran into this issue (the address was different, but otherwise same) on version 1.19. I was able to recover by updating to the latest version of the UEFI firmware (specifically, 1.27). I did not have to re-install or otherwise touch my OS. I did clear out the files related to the old version first, which may or may not make a difference.

TheMindVirus, 2021-05-31:
I did a bit of investigation into what I now affectionately call "SyncEx", it turns out this is an exception specific to Aarch64 and it can occur at different exception levels (EL1/EL2/EL3) depending on which handler was called in the Interrupt/Exception Vector Table.

It is most likely happening at EL3 due to an unaligned memory/array access in C which is using LDUR/STUR and the pi is not supporting reads/writes which are not aligned to 16 bytes (this is a cheap optimisation for speed of copying).

It could also happen for a variety of other reasons with other (compiled and assembled but otherwise unimplemented) CPU instructions, and as you saw it can affect code loaded to any address.

Some compiler options to gcc may fix the issues by telling the compiler not to use certain instructions e.g. -O2 -mstrict-align but is all a bit experimental at this stage.

michelcve, 2021-06-26:
I got a `Synchronous Exception at 0x0000000037001468` today, on a USB stick I use to provision my cluster. It was running fine for the previous 20 installs, but then it just stopped working, and getting this error as soon as I boot it.

Copying over a known working EFI didn't help.

TheMindVirus, 2021-06-26:
The error happens when the Pi is trying to do a misaligned read/write or when the NV-Variables aren’t persisting.

It either means your USB stick is faulty or that the MBR has become misaligned with the start of the disk somehow.

Reformatting the disk with a new MBR usually fixes it if a reshuffling of the files in the boot partition hasn’t worked. Unfortunately that may also mean reflashing the OS.

michelcve, 2021-06-26:
Yeah, I figured that was needed. Just mentioned it here to let everyone know that this is still happening.

Thanks though!

dcasota, 2021-12-30:
> I got a `Synchronous Exception at 0x0000000037001468` today, on a USB stick I use to provision my cluster. It was running fine for the previous 20 installs, but then it just stopped working, and getting this error as soon as I boot it.
> 
> Copying over a known working EFI didn't help.

+1 `Synchronous Exception at 0x0000000037001468`  on a continuously running Rpi4 system with fw 1.27. After updating the sdcard to fw 1.32, it booted flawlessly again.

eddcreativity, 2022-01-07:
1st wave testing:
RPi 3 model B, ARM Cortex-A53 1.20 GHz, 1024 MB RAM.
ISO : debian-11.2.0-arm64-netinst.iso
testing UEFI: RPi3_UEFI_Firmware_v1.37.zip, v1.36,  v1.35 ,v1.34,v1.33, v1.32, v1.31 v1.30
Procedure executed: https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html
missing firmware files: brcm/brcmfmac43430-sdio.raspberrypi, 3-model-b.txt
ERROR:  Synchronous Exception at 0x0000000037001468  
Is it the ISO ? , Is it was modified to don’t work anymore for RPi 3b?

dcasota, 2022-01-19:
I've downgraded a while ago to fw 1.31 because of the datetime issue #198.
fw 1.31 worked flawlessly. Having changed the media from one usb stick to one with windows 11 arm64 on it, the issue ```Synchronous Exception at 0x0000000037001468``` came back again. Bios reset didn't help.
Reburned the micro sd card with fw 1.31 as workaround.

missing firmware files: brcm/brcmfmac - this happens on Pi4 fw1.31, too.

tkoeller-max, 2023-10-01:
I also encountered this problem after an upgrade from v1.34 to v1.35 on a raspi-400. I downgraded the machine back to v1.34, and I am fine again.

commit d8a55b01 (2021-01-06): Update to EDK2 latest

* Closes #78
* Closes #82
* Closes #97