pftf/RPi4 #99: Hang with NetBSD 9.0

samerhaj, 2020-09-23:
Booting NetBSD 9.0 on RPi4 8GB model with 1.19 firmware, hangs during boot

```
>> NetBSD/evbarm efiboot (arm64), Revision 1.13 (Fri Feb 14 00:06:28 UTC 2020)
Press return to boot now, any other key for boot prompt
booting netbsd - starting in 0 seconds.
5395864+2481752+2742772+1526676 [517055+703080+492767]=0xfb5298
[   1.0000000] Copyright (c) 1996, 1997, 1998, 1999, 2000, 2001, 2002, 2003, 2004, 2005,
[   1.0000000]     2006, 2007, 2008, 2009, 2010, 2011, 2012, 2013, 2014, 2015, 2016, 2017,
[   1.0000000]     2018, 2019, 2020 The NetBSD Foundation, Inc.  All rights reserved.
[   1.0000000] Copyright (c) 1982, 1986, 1989, 1991, 1993
[   1.0000000]     The Regents of the University of California.  All rights reserved.

[   1.0000000] NetBSD 9.0 (GENERIC64) #0: Fri Feb 14 00:06:28 UTC 2020
[   1.0000000]  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/evbarm/compile/GENERIC64
[   1.0000000] total memory = 2969 MB
[   1.0000000] avail memory = 2863 MB
[   1.0000000] armfdt0 (root)
[   1.0000000] simplebus0 at armfdt0: Raspberry Pi Foundation Raspberry Pi 4 Model B
[   1.0000000] simplebus1 at simplebus0
[   1.0000000] acpifdt0 at simplebus0
[   1.0000000] acpifdt0: using EFI runtime services for RTC
[   1.0000000] ACPI: RSDP 0x0000000033AA0000 000024 (v02 RPIFDN)
[   1.0000000] ACPI: XSDT 0x0000000033A90000 000064 (v01 RPIFDN RPI4     00000200      01000013)
[   1.0000000] ACPI: FACP 0x00000000339E0000 000114 (v06 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: DSDT 0x0000000033990000 002402 (v05 RPIFDN RPI      00000002 INTL 20180105)
[   1.0000000] ACPI: CSRT 0x0000000033A00000 000169 (v00 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: DBG2 0x00000000339F0000 000061 (v00 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: GTDT 0x00000000339D0000 000068 (v03 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: APIC 0x00000000339C0000 000184 (v05 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: PPTT 0x00000000339B0000 000184 (v02 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: SPCR 0x00000000339A0000 000050 (v02 RPIFDN RPI4     00000200 EDK2 00000300)
[   1.0000000] ACPI: BGRT 0x0000000033950000 000038 (v01 INTEL  EDK2     00000002      01000013)
[   1.0000000] ACPI: 1 ACPI AML tables successfully acquired and loaded
[   1.0000000] acpi0 at acpifdt0: Intel ACPICA 20190405
[   1.0000000] cpu0 at acpi0: Cortex-A72 r0p3 (Cortex V8-A core)
[   1.0000000] cpu0: package 0, core 0, smt 0
[   1.0000000] cpu0: IC enabled, DC enabled, EL0/EL1 stack Alignment check enabled
[   1.0000000] cpu0: Cache Writeback Granule 16B, Exclusives Reservation Granule 16B
[   1.0000000] cpu0: Dcache line 64, Icache line 64
[   1.0000000] cpu0: L1 48KB/64B 3-way read-allocate PIPT Instruction cache
[   1.0000000] cpu0: L1 32KB/64B 2-way write-back read-allocate write-allocate PIPT Data cache
[   1.0000000] cpu0: L2 1024KB/64B 16-way write-back read-allocate write-allocate PIPT Unified cache
[   1.0000000] cpu0: revID=0x0, PMCv3, 4k table, 64k table, 16bit ASID
[   1.0000000] cpu0: auxID=0x10000, FP, CRC32, NEON, rounding, NaN propagation, denormals, 32x64bitRegs, Fused Multiply-Add
[   1.0000000] cpu1 at acpi0: Cortex-A72 r0p3 (Cortex V8-A core)
[   1.0000000] cpu1: package 0, core 1, smt 0
[   1.0000000] cpu2 at acpi0: Cortex-A72 r0p3 (Cortex V8-A core)
[   1.0000000] cpu2: package 0, core 2, smt 0
[   1.0000000] cpu3 at acpi0: Cortex-A72 r0p3 (Cortex V8-A core)
[   1.0000000] cpu3: package 0, core 3, smt 0
[   1.0000000] gic0 at acpi0: GIC
[   1.0000000] armgic0 at gic0: Generic Interrupt Controller, 256 sources (247 valid)
[   1.0000000] armgic0: 16 Priorities, 224 SPIs, 7 PPIs, 16 SGIs

```

Biswa96, 2020-09-23:
From USB or SD card?

samerhaj, 2020-09-23:
USB. From the image provided at EVBARM\BINARY\GZIMG\ARM64.IMG_GZ in the 9.0 ISO

rwmjones, 2020-09-24:
Is there a blank SD card in the SD slot?  It's not used, but needed (issue #37).

tbalbers, 2020-10-21:
Hi guys,
I'm seeing this exact behaviour when booting off a 32GB sd-card on a rpi4
I've tried with v.1.19 and v.1.20 and netbsd images:
http://www.armbsd.org/arm/netbsd-9/202010160010Z/NetBSD-9-aarch64-202010160010Z-generic.img.gz
https://nycdn.netbsd.org/pub/NetBSD-daily/netbsd-9/latest/evbarm-aarch64/binary/gzimg/arm64.img.gz

In both cases I replaced all files and folders in the msdos-partition except the EFI folder with what's in the firmware  gz file from here.

Both just hang at the exact same place that OP specifies.

Any suggestions are appreciated, thanks.

/tony

samerhaj, 2021-07-13:
I am still seeing this with latest UEFI (1.28) and NetBSD 9.2 I am booting the RPI_EFI from the USB, and I have a blank 32GB SD card inserted. Still hangs at the same place

samerhaj, 2021-07-13:
Support for the RPi4 is in NetBSD current, not 9. Testing from http://nycdn.netbsd.org/pub/NetBSD-daily/HEAD/latest/, everything works!

Closing this issue

closed 2021-07-13 by samerhaj