pftf/RPi4 #102: Provide some way to set/preserve variables directly on SD card

bjencks, 2020-10-01:
I normally run my pis without a console attached, and only attach a serial console for debugging and setup. Unfortunately, that's the only way to modify firmware variables now, which means I have to do it every time I replace with a newer firmware. Either a tool to set the variables in the firmware image from Linux or a way to copy variables from one version to another, would let me do in place upgrades much more easily.

garybuhrmaster, 2020-11-20:
Appears to be related to: https://github.com/pftf/RPi4/issues/6

bjencks, 2020-11-20:
Related, yes, but not the same -- I don't need to be able to make these changes while the system is running; I do need to be able to make them directly on the files, rather than through the firmware interface.