pftf/RPi4 #106: How to boot Raspberry Pi 4 UEFI Firmware from PXE?

lygstate, 2020-11-01:
By having this feature, we could have two step boot procedure, first booting the UEFI Firmware,
Then we can boot everything else, and we doesn't need the 
SD card anymore.

samerhaj, 2020-11-16:
Check the instructions here for stateless boot from PXE (for ESXi-Arm). You can extend the instructions to other OSes

andysan, 2021-02-16:
Another option is to use iSCSI for the root file system. I have been using that successfully on a PI for a few days. Unfortunately, iSCSI was dropped in 1.22, but will likely be re-added in the next build (see #125). For simple setups (especially if you have a NAS that already supports running as an iSCSI target), that might be a lot easier to configure that PXE (custom DHCP options, TFTP server, etc.). If you configure the iSCSI initiator (~client) in UEFI correctly, the remote disk shows up like an ordinary local disk at boot.

samerhaj, 2021-04-16:
The information in this issue seem to be sufficient. I will go ahead and close the issue. If you still have problems, please re-open or open a new issue.

closed 2021-04-16 by samerhaj