pftf/RPi4 #107: FDT support broken in 1.20 and 1.19

andreiw, 2020-11-04:
The FDT header is nulled out or corrupted in 1.20 and 1.19. Location is as per config.txt. Tested on Pi 4B 8GiB.

1.18 is fine.

Shell> dmem 0x1f0000
Memory � 00000000001F0000 200 Bytes
  001F0000: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  001F0010: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  001F0020: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  001F0030: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  001F0040: 00 00 00 00 00 00 00 00-00 00 00 01 00 00 00 00  *................*
  001F0050: 00 00 00 03 00 00 00 23-00 00 00 00 72 61 73 70  *.......#....rasp*
  001F0060: 62 65 72 72 79 70 69 2C-34 2D 6D 6F 64 65 6C 2D  *berrypi,4-model-*

andreiw, 2020-11-04:
This actually might be a bug with the VPU firmware.

I took 1.18 and then the latest RPI_EFI.FD.

FDT header is okay:
Shell> 
Shell> dmem 1f0000
Memory Address 00000000001F0000 200 Bytes
  001F0000: D0 0D FE ED 00 00 BE B1-00 00 00 48 00 00 AE 74  *...........H...t*
  001F0010: 00 00 00 28 00 00 00 11-00 00 00 10 00 00 00 00  *...(............*
  001F0020: 00 00 10 3D 00 00 AE 2C-00 00 00 00 00 00 00 00  *...=...,........*
  001F0030: 00 00 00 00 00 00 10 00-00 00 00 00 00 00 00 00  *................*
  001F0040: 00 00 00 00 00 00 00 00-00 00 00 01 00 00 00 00  *................*
  001F0050: 00 00 00 03 00 00 00 08-00 00 0F DD 3B 40 00 00  *............;@..*
  001F0060: 04 C0 00 00 00 00 00 03-00 00 00 11 00 00 0F CF  *................*
  001F0070: 31 30 30 30 30 30 30 30-39 33 61 32 35 63 62 31  *1000000093a25cb1*

pbatard, 2020-11-04:
I'm pretty sure this is the same issue as pftf/RPi3#22 and by extension raspberrypi/firmware#1445.

Since we're mostly using ACPI, I haven't bothered updating the RPI4 firmware, mostly because I'm waiting on https://edk2.groups.io/g/devel/message/65732 to be integrated before I release another update.

andreiw, 2020-11-04:
If there's arbitrary corruption of the ranges occupied by RPI_EFI.FD and the FDT blob, then this may be the root cause behind https://github.com/pftf/RPi4/issues/104  and https://github.com/pftf/RPi4/issues/97

closed 2020-11-13 by pbatard

einsteinagogo, 2021-01-06:
is this fixed now in 1.21 ?

samerhaj, 2021-01-06:
it seems to be, from the release notes of 1.21 :
"Fix Device Tree corruption due to a bug in start4.elf [See raspberrypi/firmware#1445, raspberrypi/firmware#1489]"

pbatard, 2021-01-06:
Again, the issue reported here was the one from raspberrypi/firmware#1445 which has long been fixed.
We expect #97 to be a byproduct of the firmware file getting corrupted during the saving of the NV variables, which was fixed in today's release.

einsteinagogo, 2021-01-06:
ah, excellent - I'll test later.

commit 8b4bb587 (2020-11-13): Update to EDK2 latest

* Closes #107