pftf/RPi4 #109: Raspberry PI 400 does not boot.

nsa666666, 2020-11-05:
The UEFI-Firmware does not boot on Raspberry PI 400. It shows only a green dash on screen.

Here is the boot log:

```
  recover4.elf not found (6)
recovery.elf not found (6)
HUB [01:00] 2.16 000000:01 init port 4 speed 2
DEV [03:01] 2.00 000004:01 class 0 VID 04d9 PID 0007
Read start4.elf bytes  2283936 hnd 0x000007ff hash 'ddddf81164f250ad'
Read fixup4.dat bytes     5422 hnd 0x00000032 hash 'fdfbb390f4a2c93c'
0x00c03130 0x00000000 0x0000001f
MEM GPU: 76 ARM: 947 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0

MESS:00:00:05.200720:0: bfs_xhci_retrieve 3ee78080
MESS:00:00:05.205136:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.208005:0: brfs: File read: 196 bytes
MESS:00:00:05.264980:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.273786:0: brfs: File read: 196 bytes
MESS:00:00:05.768260:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.775512:0: *** Restart logging
MESS:00:00:05.819660:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.825603:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.831076:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:05.841492:0: dtb_file 'bcm2711-rpi-400.dtb'
MESS:00:00:05.843618:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:05.850761:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:05.854457:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x1f0000 size 0xb99c
MESS:00:00:05.873411:0: brfs: File read: 47516 bytes
MESS:00:00:05.959786:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.962346:0: brfs: File read: 196 bytes
MESS:00:00:05.966850:0: Failed to load overlay 'miniuart-bt'
MESS:00:00:05.972145:0: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
MESS:00:00:05.986908:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:06.939047:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:06.941508:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:06.947332:0: No compatible kernel found
MESS:00:00:06.951835:0: Device tree loaded to 0x1f0000 (size 0xbe71)
MESS:00:00:06.957952:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:06.966993:0: uart: Baud rate change done...
MESS:00:00:06.969010:0:NOTICE:  BL31: v2.3():v2.3
NOTICE:  BL31: Built : 10:40:51, Apr 21 2020
UEFI firmware (version UEFI Firmware v1.20 (DEBUG) built at 14:05:52 on Sep  1 2020)
Board Rev: 0xC03130
RAM < 1GB: 0x00000000 (Size 0x3B300000)
VideoCore: 0x3B300000 (Size 0x04C00000)
Total RAM: 0x100000000
FD:
	PhysicalBase: 0x0
	VirtualBase: 0x0
	Length: 0x1D0000
FD Variables:
	PhysicalBase: 0x1D0000
	VirtualBase: 0x1D0000
	Length: 0x20000
Flattened Device Tree:
	PhysicalBase: 0x1F0000
	VirtualBase: 0x1F0000
	Length: 0x10000
System RAM < 1GB:
	PhysicalBase: 0x200000
	VirtualBase: 0x200000
	Length: 0x3B100000
GPU Reserved:
	PhysicalBase: 0x3B300000
	VirtualBase: 0x3B300000
	Length: 0x4C00000
SoC Reserved (27xx):
	PhysicalBase: 0xFC000000
[... 1274 lines cut ...]
InstallProtocolInterface: D3B36F2C-D551-11D4-9A46-0090273FC14D 0
InstallProtocolInterface: D3B36F2D-D551-11D4-9A46-0090273FC14D 0
InstallProtocolInterface: 9042A9DE-23DC-4A38-96FB-7ADED080516A 36F8A4F0
Terminal - Mode 0, Column = 80, Row = 25
Terminal - Mode 1, Column = 80, Row = 50
Terminal - Mode 2, Column = 100, Row = 31
[2J[04D[=3h[2J[09DInstallProtocolInterface: 387477C1-69C7-11D2-8E39-00A0C969723B 35F25340
InstallProtocolInterface: DD9E7534-7762-4698-8C14-F58517A625AA 35F25428
InstallProtocolInterface: 387477C2-69C7-11D2-8E39-00A0C969723B 35F25358
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 35F25918
InstallProtocolInterface: D3B36F2C-D551-11D4-9A46-0090273FC14D 0
InstallProtocolInterface: D3B36F2D-D551-11D4-9A46-0090273FC14D 0
[0m[35m[40mInstallProtocolInterface: D3B36F2B-D551-11D4-9A46-0090273FC14D 0
XhcCreateUsb3Hc: Capability length 0x20
XhcCreateUsb3Hc: HcSParams1 0x5000420
XhcCreateUsb3Hc: HcSParams2 0xFC000031
XhcCreateUsb3Hc: HcCParams 0x2841EB
XhcCreateUsb3Hc: DBOff 0x100
XhcCreateUsb3Hc: RTSOff 0x200
XhcCreateUsb3Hc: UsbLegSupOffset 0x0
XhcCreateUsb3Hc: DebugCapSupOffset 0x260
XhcSetBiosOwnership: called to set BIOS ownership
XhcResetHC!
ASSERT [XhciDxe] /home/appveyor/projects/rpi4/edk2/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c(2018): !((((BOOLEAN)(((XhcReadOpReg ((Xhc), (0x0004))) & ((0x00000800))) == ((0x00000800))))))
```

Biswa96, 2020-11-06:
C'mon! Give them some time to breath, it's released some days ago!

jeroenbrons, 2020-11-08:
i let it run for a while on the green cursor (i didnt keep track of how long) and it suddenly started booting (at least the UEFI part). ill give it a bootable AARCH64 OS (Linux-based) and report if it boots. if it boots, ill let my other system log  the UART for debugging

jeroenbrons, 2020-11-08:
it took around 60 min to start, and it rebooted  around the time it normally started booting the OS

ader1990, 2020-11-09:
@jeroenbrons after it booted, all the things worked like on the pi 4 b?

jeroenbrons, 2020-11-09:
no, it got back to the raspi rainbow-boot screen and the green cursor

nsa666666, 2020-11-13:
Seems to be some watchdog timer. It does not realy boot.

darkain, 2020-11-13:
Still not booting correctly here on the v1.21 release.

It sits on the green cursor like everyone else reported for ~25 seconds, and then shows the UEFI splash screen. USB is never activated, so not able to boot past this point or enter into the UEFI menu.

After the 5-second countdown on the splash screen, the whole device reboots.

Tested with 2 different SD cards, same results. Both tested on a Pi4-4GB, and it entered UEFI instantly with USB working there.

jeroenbrons, 2020-11-13:
i compiled a debug-binary without XHCI to test (because it stalls for a while there), keyboard doesnt work, but i can get into the menu with the UART (al least with aforementioned custom build)

nsa666666, 2020-11-14:
It is at least some improvemend. It comes to the UEFI splash screen, but keyboard does not work and then it just reboots.

worproject, 2020-11-14:
@jeroenbrons an USB-C OTG adapter should work.

robstercork, 2020-11-14:
I have a similar issue on Pi4-8GB. USB Keyboard input is ignored at the splash screen. The same SDcard and keyboards (I've tried several) work fine if the SDcard is reimaged with stock pi OS.

darkain, 2020-11-14:
> @jeroenbrons an USB-C OTG adapter should work.

RIGHT! I always forget that the Pi's power port is OTG!

I tried that just now. Obviously, still has the ~25 second stall out, but was able to get it to somewhat boot from USB, and have a working external keyboard.

The 1800MHz clock speed is recognized. However, the 4GB of RAM is wonky. UEFI reports it in menu, but only shows 3GB to the OS. I know this is normal and usually configurable, however, the configuration option is inaccessible on this board.

The Pi 400 also shows up as "Unknown Raspberry Pi Model"

[image]

[image]

[image]

[image]

jimerickson, 2020-11-17:
i am also affected by this bug. my rpi-400 boots to uefi raspberry pi logo but can't enter setup as the keyboard does not respond. so it just keeps rebooting.

samerhaj, 2020-11-17:
> i am also affected by this bug. my rpi-400 boots to uefi raspberry pi logo but can't enter setup as the keyboard does not respond. so it just keeps rebooting.

To clarify. This is not a bug. This is new HW bringup/support. The RPi 400 has a newer SoCs revision than the RPi4, and requires changes to how the UEFI FW initializes the PCIe and XHCI subsystems for it to work properly

jimerickson, 2020-11-17:
ok sorry and thanks for the explanation!

samerhaj, 2020-11-18:
@andreiw made some progress and reported an issue in the RPi FW https://github.com/raspberrypi/firmware/issues/1495 . Adding here to link the two issues

tuxevil, 2020-12-24:
> To clarify. This is not a bug. This is new HW bringup/support. The RPi 400 has a newer SoCs revision than the RPi4, and requires changes to how the UEFI FW initializes the PCIe and XHCI subsystems [...]

Could it happens that my Rpi4 8GB be using a new SoC revition too? Couse i have the same problem i see here

samerhaj, 2020-12-24:
Yes it is possible that the new SoC revision is being used in some newer RPi4 models, in addition to the CM4 and RPi400.

tuxevil, 2020-12-24:
thanks for replying @samerhaj

Googulator, 2020-12-26:
> Yes it is possible that the new SoC revision is being used in some newer RPi4 models, in addition to the CM4 and RPi400.

In fact, it's not just possible, but pretty much confirmed.

http://www.raspberry-pi-case.com/pictures/rpi4b/RPi4B_8GB.jpg

The text on the die isn't readable, but the location of the corner dot matches the CM4 and 400. Unless Broadcom is making B0 silicon with the new markings specifically for the "plain" Pi 4, this is C0 silicon.

Note that the 3GB PCIE address space bug (and possibly also the 64-bit addressing bug) is fixed in the C0 revision, so PCIE can actually access the whole 8GB address space now. The 1GB limit still applies to legacy BCM2835 peripherals.

tuxevil, 2020-12-30:
> In fact, it's not just possible, but pretty much confirmed.
>
> http://www.raspberry-pi-case.com/pictures/rpi4b/RPi4B_8GB.jpg
>
> The text on the die isn't readable, but the location of the corner dot [...]

So I guess that when this be fixed for CM4 and 400 It will be also fixed for new 4 8gb models right?

Googulator, 2020-12-30:
And Now I'm not sure what to think.

https://www.pi-shop.ch/media/catalog/product/cache/1/image/650x/040ec09b1e35df139433887a97daa66f/f/a/far_1024x1024.jpg

https://pbs.twimg.com/media/EZ1FLZcVAAIZ5V4.jpg:orig

This is clearly the new marking style on the chip, but it's marked 2711ZPKFSB06**B0**T. So, either Broadcom is in fact still producing B0 silicon alongside the newer C0 (to ensure full backwards compatibility?), or the Pi Foundation treated the "B0T" as part of the serial number, and asked for

browsingman, 2021-01-26:
Well, but what about Raspberry PI 400 not booting with UEFI bios? Still an unsolved issue ... any progress on that?

jlinton, 2021-01-28:
I think we are waiting on: https://github.com/raspberrypi/firmware/issues/1518#issuecomment-765307139

tuxevil, 2021-02-14:
as weird as it may sounds, removing the line "gpu_mem=16" worked for me

TheNeoStormZ, 2021-02-15:
> 
> 
> as weird as it may sounds, removing the line "gpu_mem=16" worked for me

from where did you remove that? And what does work for you now?

jeroenbrons, 2021-02-15:
i couldn't find a line "gpu_mem=16" (or any other gpu_mem line), but it started trying to PXE-boot (there was no ARM64-UEFI bootable device attached, i will try with one soon)

tuxevil, 2021-02-15:
> > as weird as it may sounds, removing the line "gpu_mem=16" worked for me
> 
> from where did you remove that? And what does work for you now?

well, in every youtube video i found, they are suggesting to add this line at the end of config.txt to "improve" memory usage
and that was my problem, after i removed it, i could install ESXI and use it

jeroenbrons, 2021-02-19:
i put the current ZIP in the boot-partition of a Windows-on-Raspberrypi SD-card and let it run, it boots (my PI400 is running the sysprep phase as i'm writing this)

nsa666666, 2021-02-19:
It actually boots! But USB and Keyboard does not work.
I'll open an new issue.

closed 2021-02-19 by nsa666666