pftf/RPi4 #114: Re-add connect all devices on an ordinary boot

lory696, 2020-11-20:
When the rpi4 uefi firmware boots normally, it does not connect every device, but only the one from which was set to boot from and the sd-card. In some situations, this is problematic. For example, if you want to boot grub from a specific drive (or from pxe) and then chainload another bootloader in another drive. The commit that defined this behavior is [this]( https://github.com/tianocore/edk2-platforms/commit/c8000ecccc83b728baf04ced2fedb870bc3bc1b3).

samerhaj, 2020-11-25:
Referencing EDK2-devel [patch](https://edk2.groups.io/g/devel/topic/74668022#60720) and [series](https://edk2.groups.io/g/devel/message/60717)

@ardbiesheuvel

ardbiesheuvel, 2020-11-27:
I wonder if we should something like LOAD_OPTION_FORCE_RECONNECT for normal boot options as well.

In the mean time, we could simply add a menu option that restores the old behavior, but I would prefer to keep that as a configurable option.

samerhaj, 2020-12-09:
There is an existing UEFI protocol: EFI_BOOT_MANAGER_POLICY_PROTOCOL that allows for setting the BDS connect policy to things like: Connect consoles, connect network devices, and connect all. This allows selecting different levels of "fast boot" (at a minimum connect only consoles, and connect all, with maybe a granularity of connect network)

There is an existing EDK2 implementation of MdeModulePkg/Universal/BootManagerPolicyDxe which publishes the UEFI protocol to configure the boot policy. However, there is no UI/Shell command to configure it.

A solution is to add a user config option to select the policy (with a PCD to set the platform default), similar to the Console Preference Selection driver (EmbeddedPkg/Drivers/ConsolePrefDxe):

* UI: possibly add to MdeModulePkg/Library/BootMaintenanceManagerUiLib
* Shell: possibly add to ShellPkg/Library/UefiShellBcfgCommandLib

samerhaj, 2021-03-30:
For reference, this issue is also mentioned as one of two issues impacting some automated deployment/provisioning scenarios (using MAAS) : https://discourse.maas.io/t/raspberry-pi-4-provisioning-and-kvm-pod-setup/3607

> After the installation of ubuntu, maas sends to the client a different configuration file, that tells to scan every device to find the bootloader. However, when the rpi4 uefi firmware boots normally, it does not attach every device, but only the one from which was set to boot from. So, the first reboot after the deploy step fails, because the os bootloader cannot be found. It is possible to avoid this by pressing esc to enter the uefi setup and then press continue. By doing this all device are connected and the os can be booted, but you need to do it manually every time. I opened an issue 9.

samerhaj, 2021-04-12:
Please see https://github.com/pftf/RPi4/issues/144 

There is already a patch sent to EDK2 list (not merged yet). Please try to confirm it fixes this issue

lory696, 2021-04-13:
I just tested it. I confirm that the patch fixes this issue.

sunnywang-arm, 2021-04-16:
Close this issue because @ardbiesheuvel already pushed the fix and the fix was already verified by @lory696.  Thanks, guys.

closed 2021-04-16 by sunnywang-arm