pftf/RPi4 #117: Path chosen to execute UEFI shell impact the map

pelamlio, 2020-11-30:
Upon booting the UEFI from USB, the path used to launch the UEFI Shell impact the map.

At the splash screen
- F1 (shell) -> the map is not complete
- ESC -> Boot Manager -> UEFI Shell -> the map is correct

based on Raspberry Pi 4 8GB, UEFI Firmware v1.21

samerhaj, 2020-12-20:
When booting from the first path, try typing this at the shell:

```
connect -r
map -r
```

samerhaj, 2021-04-12:
This may be related to https://github.com/pftf/RPi4/issues/144

A patch has been sent to EDK2 (not merged yet). Please try and confirm if it fixes the issue

sunnywang-arm, 2021-04-16:
@samerhaj 
Yeah, just tried and confirmed this issue got fixed by the fix for #144 as well. 
I successfully reproduced the issue with the Boot Policy Fast Boot setting, and then the issue disappears with Boot Policy Full discovery setting. 

@pelamlio 
Since the fix for #144 below already got pushed, I closed this issue with my verification result above. If you still see this issue, feel free to re-open this issue.  
https://github.com/tianocore/edk2-platforms/commit/efdc159ef7c9f15581a0f63d755a1530ff475156

closed 2021-04-16 by sunnywang-arm